paukstelis/octoprint_deploy #1: Trouble setting up without serial #

4jesse, 2021-02-16:
I'm tring to setup several printers on a NUC with 4 USB ports. The first one (PRINTER1) is detected with its serial number, but the next one (PRINTER2) is not finding a serial # and is showing XXXXXXXXXX. 

I edited the /etc/udev/rules.d/99-octoprint.rules file to read:
```
SUBSYSTEM=="tty", ATTRS{serial}=="A506VOOE", SYMLINK+="octo_Printer1"
SUBSYSTEM=="tty", ATTRS{devpath}=="1-0:1.0", SYMLINK+="octo_Printer2"
```
_(I also tried it with 2-0:1.0)_

When viewing the Octoprint page for PRINTER1 (:5001) it shows /dev/octo_Printer1 under the Serial Port. When I view the page for PRINTER2 (:5002), it doesn't show a /dev/octo_Printer2 (or anything with ...octo_PrinterX. It does, however work on /dev/USB0.

Here is the full results from my dmesg:

```
octouser@OctoNUC:~/OctoPrint/bin$ dmesg
[    0.000000] Linux version 5.8.0-43-generic (buildd@lcy01-amd64-018) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #49~20.04.1-Ubuntu SMP Fri Feb 5 09:57:56 UTC 2021 (Ubuntu 5.8.0-43.49~20.04.1-generic 5.8.18)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-43-generic root=UUID=a120d476-91a3-42f8-83f6-2b6af49bba43 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x00000000774b5fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000774b6000-0x00000000795d7fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000795d8000-0x00000000795f2fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000795f3000-0x0000000079652fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079653000-0x00000000799f5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000799f6000-0x0000000079d78fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079d79000-0x0000000079d79fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079d7a000-0x0000000079da3fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079da4000-0x000000007a9defff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a9df000-0x000000007a9e0fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007a9e1000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe044fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe900000-0x00000000fe902fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x724f6018-0x72506057] usable ==> usable
[    0.000000] e820: update [mem 0x724f6018-0x72506057] usable ==> usable
[    0.000000] e820: update [mem 0x724e7018-0x724f5057] usable ==> usable
[    0.000000] e820: update [mem 0x724e7018-0x724f5057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000003f000-0x000000000003ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000040000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000010000000-0x0000000012150fff] reserved
[... 969 lines cut ...]
[    9.055099] wlp2s0: send auth to 8e:dc:96:88:33:1c (try 1/3)
[    9.156883] wlp2s0: send auth to 8e:dc:96:88:33:1c (try 2/3)
[    9.158943] wlp2s0: authenticated
[    9.160872] wlp2s0: associate with 8e:dc:96:88:33:1c (try 1/3)
[    9.164170] wlp2s0: RX AssocResp from 8e:dc:96:88:33:1c (capab=0x1511 status=0 aid=1)
[    9.165501] wlp2s0: associated
[    9.210875] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[    9.215782] wlp2s0: Limiting TX power to 27 (30 - 3) dBm as advertised by 8e:dc:96:88:33:1c
[   12.657374] rfkill: input handler disabled
[   13.543003] kauditd_printk_skb: 30 callbacks suppressed
[   13.543005] audit: type=1400 audit(1613501267.243:41): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=1483 comm="snap-confine" capability=4  capname="fsetid"
[   17.352443] audit: type=1326 audit(1613501271.051:42): auid=1000 uid=1000 gid=1000 ses=2 subj=snap.snap-store.ubuntu-software pid=1483 comm="snap-store" exe="/snap/snap-store/518/usr/bin/snap-store" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f79b1398639 code=0x50000
[   17.853481] audit: type=1107 audit(1613501271.555:43): pid=601 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/NetworkManager" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.12" pid=1483 label="snap.snap-store.ubuntu-software" peer_pid=602 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   19.257094] audit: type=1107 audit(1613501272.959:44): pid=601 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.9" pid=1483 label="snap.snap-store.ubuntu-software" peer_pid=620 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   19.258205] audit: type=1107 audit(1613501272.959:45): pid=601 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" mask="send" name=":1.9" pid=1483 label="snap.snap-store.ubuntu-software" peer_pid=620 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   19.288406] audit: type=1107 audit(1613501272.987:46): pid=601 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.9" pid=1483 label="snap.snap-store.ubuntu-software" peer_pid=620 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   19.289234] audit: type=1107 audit(1613501272.991:47): pid=601 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" mask="send" name=":1.9" pid=1483 label="snap.snap-store.ubuntu-software" peer_pid=620 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   19.836968] audit: type=1400 audit(1613501273.539:48): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/etc/PackageKit/Vendor.conf" pid=1483 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[ 7262.583892] perf: interrupt took too long (2508 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
```

paukstelis, 2021-02-16:
Have you reloaded the rules since you edited your udev rules file?

try:
`sudo udevadm control --reload-rules`
`sudo udevadm trigger`

You can then check to see what you have in /dev with: `ls /dev/octo*`

4jesse, 2021-02-17:
Still only Printer1. 
From my dmesg post, am I using the correct devpath attribute?

paukstelis, 2021-02-17:
It looks like it, but if  you could unplug/plug the printer again in the same location, immediately do a dmesg and just take the last 20 or so lines from dmesg it might be more instructive.

paukstelis, 2021-02-17:
You might be better off using idProduct and idVendor.

SUBSYSTEM=="tty", ATTRS{idVendor}=="1a86", ATTR{idProduct}=="7523", SYMLINK+="octo_Printer2"

Do the udevadm commands again. Then check what is in the dev directory.

4jesse, 2021-02-17:
No joy. I tried adding Printer3 (also not detecting the serial#) - it came up as 3-1,1.0 but has the exact same idVendor and idProduct as Printer2. I tried both the {devpath} option as well as the suggestion above and it still won't create the SymLink. I did the udevadm commands again and rebooted the machine. No joy.

These are both original Creality boards. I'm going to try to update the firmware and see if anything changes.

paukstelis, 2021-02-17:
Hmm...I don't use devpath for my printers since they all have serial numbers. For cameras I have used KERNELS, which you could try. Something like:

SUBSYSTEM="tty",KERNELS=="1-1:1.0",SUBSYSTEMS=="usb",SYMLINK+="octo_Printer2"

4jesse, 2021-02-17:
I used USB1 and got this output:

```
octouser@OctoNUC:~$ udevadm info --attribute-walk --name=/dev/ttyUSB1

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:00/0000:00:15.0/usb1/1-2/1-2:1.0/ttyUSB1/tty/ttyUSB1':
    KERNEL=="ttyUSB1"
    SUBSYSTEM=="tty"
    DRIVER==""

  looking at parent device '/devices/pci0000:00/0000:00:15.0/usb1/1-2/1-2:1.0/ttyUSB1':
    KERNELS=="ttyUSB1"
    SUBSYSTEMS=="usb-serial"
    DRIVERS=="ftdi_sio"
    ATTRS{port_number}=="0"
    ATTRS{latency_timer}=="16"

  looking at parent device '/devices/pci0000:00/0000:00:15.0/usb1/1-2/1-2:1.0':
    KERNELS=="1-2:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="ftdi_sio"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bNumEndpoints}=="02"
    ATTRS{bInterfaceProtocol}=="ff"
    ATTRS{bInterfaceClass}=="ff"
    ATTRS{supports_autosuspend}=="1"
    ATTRS{interface}=="FT232R USB UART"
    ATTRS{authorized}=="1"
    ATTRS{bInterfaceSubClass}=="ff"

  looking at parent device '/devices/pci0000:00/0000:00:15.0/usb1/1-2':
    KERNELS=="1-2"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{bMaxPacketSize0}=="8"
    ATTRS{speed}=="12"
    ATTRS{manufacturer}=="FTDI"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{product}=="FT232R USB UART"
    ATTRS{version}==" 2.00"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{devnum}=="3"
    ATTRS{quirks}=="0x0"
    ATTRS{bcdDevice}=="0600"
    ATTRS{authorized}=="1"
    ATTRS{bmAttributes}=="a0"
    ATTRS{tx_lanes}=="1"
    ATTRS{idVendor}=="0403"
    ATTRS{maxchild}=="0"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bDeviceClass}=="00"
    ATTRS{devpath}=="2"
    ATTRS{configuration}==""
    ATTRS{ltm_capable}=="no"
    ATTRS{busnum}=="1"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPower}=="90mA"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{idProduct}=="6001"
    ATTRS{urbnum}=="16"
    ATTRS{removable}=="removable"
    ATTRS{serial}=="A506VOOE"
    ATTRS{rx_lanes}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:15.0/usb1':
    KERNELS=="usb1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{idVendor}=="1d6b"
    ATTRS{product}=="xHCI Host Controller"
    ATTRS{authorized_default}=="1"
    ATTRS{configuration}==""
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{tx_lanes}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{urbnum}=="99"
    ATTRS{maxchild}=="8"
    ATTRS{idProduct}=="0002"
    ATTRS{bmAttributes}=="e0"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{authorized}=="1"
    ATTRS{version}==" 2.00"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{quirks}=="0x0"
    ATTRS{ltm_capable}=="no"
    ATTRS{bcdDevice}=="0508"
    ATTRS{removable}=="unknown"
    ATTRS{serial}=="0000:00:15.0"
    ATTRS{speed}=="480"
    ATTRS{devpath}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{devnum}=="1"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{busnum}=="1"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{manufacturer}=="Linux 5.8.0-43-generic xhci-hcd"
    ATTRS{rx_lanes}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:15.0':
    KERNELS=="0000:00:15.0"
    SUBSYSTEMS=="pci"
    DRIVERS=="xhci_hcd"
    ATTRS{subsystem_vendor}=="0x8086"
    ATTRS{broken_parity_status}=="0"
    ATTRS{enable}=="1"
    ATTRS{class}=="0x0c0330"
    ATTRS{consistent_dma_mask_bits}=="64"
    ATTRS{revision}=="0x0b"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{local_cpus}=="f"
    ATTRS{vendor}=="0x8086"
    ATTRS{numa_node}=="-1"
    ATTRS{subsystem_device}=="0x2067"
    ATTRS{msi_bus}=="1"
    ATTRS{device}=="0x5aa8"
    ATTRS{ari_enabled}=="0"
    ATTRS{dbc}=="disabled"
    ATTRS{irq}=="126"
    ATTRS{local_cpulist}=="0-3"
    ATTRS{driver_override}=="(null)"
    ATTRS{dma_mask_bits}=="64"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""

octouser@OctoNUC:~$ 
```

paukstelis, 2021-02-17:
This should work (assuming you still have them plugged into the same usb ports):

KERNELS==""1-1:1.0",SUBSYSTEMS=="usb",SYMLINK+="octo_Printer2"
KERNELS==""1-2:1.0",SUBSYSTEMS=="usb",SYMLINK+="octo_Printer3"

I tested this on my system and it worked.

4jesse, 2021-02-17:
Thanks for all the options, but that didn't work either. I've just finished updating the firmware with TH3D and wanted to try the addnew script again, but it says "Already have an entry for Printer2. Exiting." I had already removed the two extra printers (left PRINTER1) from the 99-octoprint.rules file.

So, how do I remove a previous attempt?

paukstelis, 2021-02-17:
It is not firmware dependent (as far as I know) so you aren't likely to have any luck with that. You can plug it in and check the end of the dmesg again to see if there is a serial number, but I doubt it.

What I gave you definitely works in a general sense, so perhaps there is some issue with how the udev line is formatted. 

Removing whole instances requires deleting:
/home/x/.INSTANCE_NAME
/etc/default/INSTANCE_NAME
/etc/init.d/INSTANCE_NAME
and removing the port of that instance from /etc/octoprint_ports.

4jesse, 2021-02-18:
Yep, you're right. No change after the firmware change. Previously I did copy/paste of your KERNELS option above and it didn't work, but I JUST looked at the syntax and I should have removed an extra quote mark after KERNELS==. 
```
KERNELS==""1-1:1.0",SUBSYSTEMS=="usb",SYMLINK+="octo_Printer2"
KERNELS==""1-2:1.0",SUBSYSTEMS=="usb",SYMLINK+="octo_Printer3"
```
It's now all good - your suggestion worked great! I'm not sure why the other methods didn't work for me, but I'm thrilled you kept trying to find options for me. Thanks MUCH.

paukstelis, 2021-02-18:
Sorry about that. Copy/paste errors! Good luck!

closed 2021-02-18 by paukstelis