paukstelis/octoprint_deploy #2: Youtube comments - Connor Heffer - OctoBuntu

xHeffer, 2021-03-26:
here is my dmesg

[ 2860.622875] usb 3-1: USB disconnect, device number 7
[ 2860.623154] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 2860.623179] ch341 3-1:1.0: device disconnected
[ 2891.268746] usb 3-1: new full-speed USB device number 8 using xhci_hcd
[ 2891.417833] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 2891.417836] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 2891.417837] usb 3-1: Product: USB Serial
[ 2891.418576] ch341 3-1:1.0: ch341-uart converter detected
[ 2891.419007] usb 3-1: ch341-uart converter now attached to ttyUSB0
[ 2893.288389] systemd-journald[261]: Received client request to rotate journal.
[ 2893.449946] systemd-journald[261]: Vacuuming done, freed 0B of archived journals from /var/log/journal/88020997ada14fb190115a38d7af21af.
[ 3026.738385] usb 3-1: USB disconnect, device number 8
[ 3026.738663] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 3026.738689] ch341 3-1:1.0: device disconnected
[ 3029.191948] usb 3-1: new full-speed USB device number 9 using xhci_hcd
[ 3029.341104] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 3029.341107] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 3029.341109] usb 3-1: Product: USB Serial
[ 3029.341867] ch341 3-1:1.0: ch341-uart converter detected
[ 3029.342227] usb 3-1: ch341-uart converter now attached to ttyUSB0
[ 5921.604857] usb 3-1: USB disconnect, device number 9
[ 5921.605691] usb 3-1: failed to send control message: -19
[ 5921.606329] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 5921.606347] ch341 3-1:1.0: device disconnected
[ 5952.396723] usb 3-1: new full-speed USB device number 10 using xhci_hcd
[ 5952.545829] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 5952.545833] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 5952.545835] usb 3-1: Product: USB Serial
[ 5952.547327] ch341 3-1:1.0: ch341-uart converter detected
[ 5952.548053] usb 3-1: ch341-uart converter now attached to ttyUSB0

paukstelis, 2021-03-26:
Thanks. This has an entirely different USB/UART adapter. I will add it in and then it should work. It still doesn't have a serial number, but it should at least detect the port. I will let you know when I have the change to try out.

What printer board is this?

paukstelis, 2021-03-26:
Following commands in the octoprint_deploy directory should let you try the new version:

git pull
git checkout ch341

xHeffer, 2021-03-26:
That's the Ender 3 max board 
I'm also adding the tronxy xy2 onto it which is a chitu board

paukstelis, 2021-03-26:
If you can give me the dmesg from teh tronxy that would be useful as well.

xHeffer, 2021-03-26:
I'll sort that tommorow once I'm available also having an anycubic mega zero in there which is a very standard board 

Do I need the 99 octoprint rules file?

paukstelis, 2021-03-26:
No, the  file will get created for you.

xHeffer, 2021-03-26:
Also camera wise I tried the camera setup and it didn't recognise it either but I'll look at that tommorow

paukstelis, 2021-03-26:
Cameras are a lot trickier. That feature is very experimental.

xHeffer, 2021-03-26:
octouser@octolinux:~/octoprint_deploy$ git pull
From https://github.com/paukstelis/octoprint_deploy
 * [new branch]      ch341      -> origin/ch341
Updating 5562539..a33efb3
error: Your local changes to the following files would be overwritten by merge:
        addnew_printer.sh
        octoprint_default
        octoprint_udev
Please commit your changes or stash them before you merge.
error: The following untracked working tree files would be overwritten by merge:
        octocam_generic.service
        octoprint_generic.service
        uninstall_instance.sh
Please move or remove them before you merge.
Aborting
 

is that normal

paukstelis, 2021-03-27:
no, that's kind of weird, but you can do a: git stash;  git pull and it should resolve things.

xHeffer, 2021-03-27:
Done that managed to pull ok tried setup again but nothing? Same output

paukstelis, 2021-03-27:
Did you switch branches? `git checkout ch341`
Can you try one of the other printers to see what happens?

xHeffer, 2021-03-27:
I've tried all three none pickup anything 
ill send d msg of each one Here is anycubic

als from /var/log/journal/88020997ada14fb190115a38d7af21af.
[59651.122839] usb 1-1.2: new full-speed USB device number 13 using ehci-pci
[59651.232237] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[59651.232240] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[59651.232242] usb 1-1.2: Product: USB Serial
[59651.232672] ch341 1-1.2:1.0: ch341-uart converter detected
[59651.233567] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
[59667.324773] usb 1-1.2: USB disconnect, device number 13
[59667.324977] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[59667.325002] ch341 1-1.2:1.0: device disconnected
[59674.903478] systemd-journald[261]: Received client request to rotate journal.
[59675.166189] systemd-journald[261]: Vacuuming done, freed 0B of archived journals from /var/log/journal/88020997ada14fb190115a38d7af21af.
[59678.510713] usb 1-1.2: new full-speed USB device number 14 using ehci-pci
[59678.620083] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[59678.620086] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[59678.620088] usb 1-1.2: Product: USB Serial
[59678.621014] ch341 1-1.2:1.0: ch341-uart converter detected
[59678.622405] usb 1-1.2: ch341-uart converter now attached to ttyUSB0

here is ender max
[59678.620088] usb 1-1.2: Product: USB Serial
[59678.621014] ch341 1-1.2:1.0: ch341-uart converter detected
[59678.622405] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
[59865.928930] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[59865.929801] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[59865.978193] usb 1-1.2: USB disconnect, device number 14
[59865.979380] usb 1-1.2: failed to send control message: -19
[59865.979507] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[59865.979528] ch341 1-1.2:1.0: device disconnected
[59868.565986] usb 1-1.2: new full-speed USB device number 15 using ehci-pci
[59868.675429] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[59868.675433] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[59868.675434] usb 1-1.2: Product: USB Serial
[59868.675981] ch341 1-1.2:1.0: ch341-uart converter detected
[59868.676998] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
octouser@octolinux:~$ 

here is tronxy
[59868.565986] usb 1-1.2: new full-speed USB device number 15 using ehci-pci
[59868.675429] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[59868.675433] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[59868.675434] usb 1-1.2: Product: USB Serial
[59868.675981] ch341 1-1.2:1.0: ch341-uart converter detected
[59868.676998] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
[59897.056250] usb 1-1.2: USB disconnect, device number 15
[59897.056534] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[59897.056561] ch341 1-1.2:1.0: device disconnected
[59902.473878] usb 3-1: new full-speed USB device number 11 using xhci_hcd
[59902.622984] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[59902.622987] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[59902.622989] usb 3-1: Product: USB Serial
[59902.624441] ch341 3-1:1.0: ch341-uart converter detected
[59902.625175] usb 3-1: ch341-uart converter now attached to ttyUSB0

paukstelis, 2021-03-27:
Weird. Wish I had one of those boards to test with. The common thread here is that they are all using ch341 usb-uart.

I added a script to the ch341 branch that can help diagnose without having to run through the whole process. do a git pull and then there is a usbtesting.sh file that you should run as sudo: $ sudo ./usbtesting.sh

It probably won't work at first, but I can make some modifications to the script to try and figure it out.

xHeffer, 2021-03-27:
Email me where you are based I can send you a tronxy board I have spare?
***@***.***

xHeffer, 2021-03-27:
Plug your printer in via USB now (detection time-out in 1 min)
1-1.2:1.0
1-1.2:1.0
1-1.2:1.0
1-1.2:1.0
1-1.2:1.0
1-1.2:1.0

xHeffer, 2021-03-27:
Could it be a driver issue?

paukstelis, 2021-03-27:
That indicates it is actually working. Those boards don't have serial numbers, but it has detected the USB port it was plugged into.

I will modify the main script slightly and you can give that a shot again. It _should_ timeout and give you that usb port assignment.

git pull

sudo ./addnew_printer.sh

xHeffer, 2021-03-27:
done that this is what I get after is this correct
Do you want to use the physical USB port to assign the udev entry? If you use this any USB hubs and printers detected this way must stay plugged into the same USB positions on your machine as they are right now (y/n).y
Your printer will be setup at the following usb address:
1-1.2:1.0

Would you like to auto detect an associated USB camera (experimental; y/n)?n
Ready to write all changes. Do you want to proceed? y
Created symlink /etc/systemd/system/multi-user.target.wants/avc.service → /etc/systemd/system/avc.service.
 only creates symlink?

paukstelis, 2021-03-27:
No, it sets up the whole service. You should be able to access your instances at whatever port you setup.

xHeffer, 2021-03-27:
So from here how do I start?

xHeffer, 2021-03-27:
Don't worry got it working how may I buy you a couple coffees kind sir!

paukstelis, 2021-03-27:
Great and thanks! Bottom of this page: https://github.com/paukstelis/Octoprint-Cancelobject

xHeffer, 2021-03-27:
Perfect :)

xHeffer, 2021-04-02:
@paukstelis after adding all the non serial printers im trying to add another printer (4th printer) which is a btt skr 1.4 any clue on why its picking upp commands from another printer@?

paukstelis, 2021-04-02:
Contents of /etc/udev/rules.d/99-octoprint.rules would be useful. See if there duplicates for some reason.

paukstelis, 2021-04-02:
Along those same lines, a $ ls -la /dev/octo_* would see if you have two symlinks going to the same thing.

TwystNeko, 2021-04-30:
[ 1008.332589] usb 1-1.4: USB disconnect, device number 14
[ 1027.262845] systemd-journald[115]: Received request to rotate journal from PID 1
[ 1027.277133] systemd-journald[115]: Vacuuming done, freed 0B of archived journals from /run/log/journal/895a3f8310d74f8d95121cf8f2647016.
[ 1036.794313] usb 1-1.4: new full-speed USB device number 15 using dwc_otg
[ 1036.928937] usb 1-1.4: New USB device found, idVendor=1eaf, idProduct=0029, bcdDevice= 2.00
[ 1036.928956] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1036.928970] usb 1-1.4: Product: Maple
[ 1036.928983] usb 1-1.4: Manufacturer: LeafLabs
[ 1036.930193] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

So I'm trying to hook up a CR6-SE with a BTT CR6 board, and this is the dmesg I get.
Also won't detect an Ender 3v2 with the CH341 chipset.
Cameras get detected just fine.

closed 2022-06-09 by paukstelis