paukstelis/octoprint_deploy #3: start over

manningcustom, 2021-11-20:
I was able to setup 4 printers using this method.   only 2 of them had serial numbers that were recognized.    I remember it saying that the others I would have to make sure I used the same ports and hubs if they were removed.    

So I put everything in place in my cabinet and cleaned up all the wires and now nothing it matching up correctly.   I was able to connect to one printer but it was under another printers connection.    my genius was controlled by my cr10v2 url connection.    

Is there a way to delete all the octoprint instances and start over now that they are all where they will be for the long term.   I know I could wipe the computer and reinstall everything but I'm sure there is simpler way.

paukstelis, 2021-11-20:
Yes. You can use the uninstall script: `sudo octoprint_deploy/uninstall_instance.sh`

manningcustom, 2021-11-20:
Thanks

does that delete them all?  or one at a time?

Also, I want to use a bunch of usb cameras.   I assume I can just plug them all into a few usb hubs?   Is there a tutorial somewhere you can direct me to?   I'm assuming there are a few things I'll need to learn about linux to accomplish this part.

manningcustom, 2021-11-21:
I've deleted them and now re installing them.   I don't think it removed the port assignment.  If that is what is even going on.   the port that it is selecting now is 5007.  The next one in line after all the others.   Will I run out of ports?   Is this even a worry?

paukstelis, 2021-11-21:
Hmm...it will increment from the last present port. It doesn't really matter, however. You won't run out of ports.

manningcustom, 2021-11-21:
when I rebooted all of the machines switched places.   I have 5 setup.

paukstelis, 2021-11-21:
The only reason that could possibly happen is if the udev rules are messed up somehow. 

Let me know the output of the command: /etc/udev/rules.d/99-octoprint.rules

Also, what computer hardware are you using?

manningcustom, 2021-11-21:
SUBSYSTEM=="tty", ATTRS{serial}=="AG0JY07M", SYMLINK+="octo_sv01_1"
SUBSYSTEM=="tty", ATTRS{serial}=="AK06VR1D", SYMLINK+="octo_sv01_2"
KERNELS=="1-2.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_CR10V2"
KERNELS=="1-1.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_artillery S!"
KERNELS=="1-1.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_S1_1"
SUBSYSTEM=="tty", ATTRS{serial}=="AG0JY07M", SYMLINK+="octo_SV01_1"
SUBSYSTEM=="tty", ATTRS{serial}=="AK06VR1D", SYMLINK+="octo_SV01_2"
KERNELS=="1-2.2:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Genius_2"

I am using a HP thin client  HP t620
amd processor I think.   64 gig hard drive  8 gig  of ram.   

I have newer computers I can switch to if you think it is needed.

paukstelis, 2021-11-21:
Nope, computer is fine. The problem are the duplicates in the udev rules. I'm not sure why they wouldn't have been removed when  you did the uninstalls. They should be and I confirmed it worked on my system. We should confirm that you removed the service files too, so do a: ls /etc/systemd/system/ and post the response.

To fix this, you need to make sure there are no duplicates of either serial or KERNELS. You should be able to edit and just delete the first four lines:
sudo nano /etc/udev/rules.d/99-octoprint.rules

The one thing I notice is that the CR10V2 and Genius_2 are on the same position, so you might have a look to see which one is supposed to be there.

manningcustom, 2021-11-22:
i think i found the problem.   USB connection issues

manningcustom, 2021-11-22:
Ok I think I'm down to the last issue (other than webcams).   

The same as before.   When I reboot they switch places.    I've slowly tested every step so here is all the info.

When I was setting each of them up I rebooted after the first 2 and no problem.   I also rebooted after the 3rd.   No  problem.   I didn't  reboot again until I was done.   I have 6 printers setup.   They all connected and can be controlled.   But, each time I reboot they get mixed up.   No pattern.

Here is the output of the script you gave me earlier.   Interesting that there are only 5 instances and there are 6 printers.   

KERNELS=="1-2.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_CR10V2"
KERNELS=="1-1.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_S1_1"
SUBSYSTEM=="tty", ATTRS{serial}=="AG0JY07M", SYMLINK+="octo_SV01_1"
SUBSYSTEM=="tty", ATTRS{serial}=="AK06VR1D", SYMLINK+="octo_SV01_2"
KERNELS=="1-2.2:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Genius_2"

Again, thank you for your help.   This project is great and when I get the bugs out will make my setup great!

paukstelis, 2021-11-22:
Though it is important that you NOT use the AUTO connect. You need to make sure you specify the printer you want to connect to
[image]
.

manningcustom, 2021-11-22:
I'll try the auto connect first.   I did have that on.

manningcustom, 2021-11-22:
[image]

manningcustom, 2021-11-22:
Mon 22 Nov 2021 09:22:09 AM PST starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid

Mon 22 Nov 2021 09:40:04 AM PST starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid
1-2.1:1.0
Mon 22 Nov 2021 10:38:02 AM PST starting instance installation
Selected port is: 5003
Executable path is valid
Template path is valid
1-1.1:1.0
Mon 22 Nov 2021 10:43:23 AM PST starting instance installation
Selected port is: 5003
Executable path is valid
Template path is valid
1-1.1:1.0
Mon 22 Nov 2021 10:50:54 AM PST starting instance installation
Selected port is: 5004
Executable path is valid
Template path is valid
Serial number detected as: AG0JY07M
Mon 22 Nov 2021 10:54:49 AM PST starting instance installation
Selected port is: 5005
Executable path is valid
Template path is valid
Serial number detected as: AK06VR1D
Mon 22 Nov 2021 11:00:35 AM PST starting instance installation
Selected port is: 5006
Executable path is valid
Template path is valid

Mon 22 Nov 2021 11:04:55 AM PST starting instance installation
Selected port is: 5007
Executable path is valid
Template path is valid
1-2.2:1.0
Mon 22 Nov 2021 03:32:36 PM PST starting instance installation
Already have an entry for Genius_2. Exiting.

manningcustom, 2021-11-22:
I got all but one of them to line up by simply not using auto connect.    The last one just wouldn't connect.   So I thought I'd just delete it and try again.   Under the octoprint deploy uninstall there are only 2 listed.   The generic and the cr10.   even the ones that work are not listed.

paukstelis, 2021-11-23:
You can check to see which instances have been installed and the uninstall script will be aware of with: cat /etc/octoprint_instances

closed 2022-06-09 by paukstelis

victornpb, 2023-08-17:
> Yes. You can use the uninstall script: `sudo octoprint_deploy/uninstall_instance.sh`

It seems that this no longer exists on the current brach. What is the procedure to uninstall everything that was installed by this?

victornpb, 2023-08-17:
> `sudo octoprint_deploy/octoprint_deploy.sh remove`

ty for the quick reply

maybe thats something worth mentioning on the readme?

paukstelis, 2023-08-17:
It is there. It is also in the wiki.