paukstelis/octoprint_deploy #4: Two cameras not working

enishi1357, 2021-12-27:
Sat 25 Dec 05:40:31 GMT 2021 starting instance installation
Sat 25 Dec 05:44:11 GMT 2021 starting instance installation
Executable path is valid
Template path is valid
Sat 25 Dec 05:53:04 GMT 2021 starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid
Sat 25 Dec 05:54:57 GMT 2021 starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001
Sat 25 Dec 06:20:38 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001
Sat 25 Dec 06:32:21 GMT 2021 starting instance installation
Already have an entry for A8Plus. Exiting.
Sat 25 Dec 06:48:44 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001
Sat 25 Dec 07:44:42 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001
Sat 25 Dec 08:47:37 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Sat 25 Dec 08:52:43 GMT 2021 starting instance installation
Sat 25 Dec 09:00:59 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Sat 25 Dec 09:24:43 GMT 2021 starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001
Sat 25 Dec 09:29:21 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Sat 25 Dec 22:32:36 GMT 2021 starting instance installation
Sun 26 Dec 08:52:02 GMT 2021 starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001
Sun 26 Dec 09:17:16 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Sun 26 Dec 09:25:30 GMT 2021 starting instance installation
Selected port is: 5002
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.

enishi1357, 2021-12-27:
I reinstall the img and setup only the first instance with camera. Here is the log.

Mon 27 Dec 10:15:14 GMT 2021 starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid
WARNING!! You must setup the base profile and admin user before continuing
Mon 27 Dec 10:17:58 GMT 2021 starting instance installation
Selected port is: 5001
Executable path is valid
Template path is valid

Camera Serial Number not detected
Camera will be setup with physical USB address of .
The camera will have to stay plugged into this location.
Selected port is: 8001

paukstelis, 2021-12-27:
It appears that your camera is not being detected at all. 
ssh into your Pi, give the command `dmesg -wH`, then plug in your camera. Then, Ctrl+c and copy and paste the new lines that appeared here.

paukstelis, 2021-12-27:
If you could also use the same thing for the printers, that would also be useful. It doesn't look like it is detecting a serial number and it should also be providing the usb address of the printers in the log.

enishi1357, 2021-12-27:
pi@octopi:~ $ dmesg -wH
[Dec27 10:52] Booting Linux on physical CPU 0x0
[  +0.000000] Linux version 5.4.79-v7+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:22:33 GMT 2020
[  +0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[  +0.000000] CPU: div instructions available: patching division code
[  +0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[  +0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[  +0.000000] Memory policy: Data cache writealloc
[  +0.000000] Reserved memory: created CMA memory pool at 0x34000000, size 64 MiB
[  +0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[  +0.000000] On node 0 totalpages: 229376
[  +0.000000]   Normal zone: 2016 pages used for memmap
[  +0.000000]   Normal zone: 0 pages reserved
[  +0.000000]   Normal zone: 229376 pages, LIFO batch:63
[  +0.000000] percpu: Embedded 20 pages/cpu s49740 r8192 d23988 u81920
[  +0.000000] pcpu-alloc: s49740 r8192 d23988 u81920 alloc=20*4096
[  +0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[  +0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[  +0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=87d94902-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[  +0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[  +0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[  +0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[  +0.000000] Memory: 827700K/917504K available (9216K kernel code, 698K rwdata, 2608K rodata, 1024K init, 827K bss, 24268K reserved, 65536K cma-reserved)
[  +0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[  +0.000000] ftrace: allocating 29201 entries in 58 pages
[  +0.000000] rcu: Hierarchical RCU implementation.
[  +0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[  +0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[  +0.000000] random: get_random_bytes called from start_kernel+0x324/0x4f8 with crng_init=0
[  +0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[  +0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[  +0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[  +0.000014] Switching to timer-based delay loop, resolution 52ns
[  +0.000267] Console: colour dummy device 80x30
[  +0.000593] printk: console [tty1] enabled
[  +0.000066] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[  +0.000049] pid_max: default: 32768 minimum: 301
[  +0.000207] LSM: Security Framework initializing
[  +0.000237] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[  +0.000046] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[  +0.001476] Disabling memory control group subsystem
[  +0.000142] CPU: Testing write buffer coherency: ok
[  +0.000584] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[  +0.000969] Setting up static identity map for 0x100000 - 0x10003c
[  +0.000221] rcu: Hierarchical SRCU implementation.
[  +0.000706] smp: Bringing up secondary CPUs ...
[  +0.001144] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[  +0.001274] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[  +0.001161] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[  +0.000154] smp: Brought up 1 node, 4 CPUs
[  +0.000089] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[  +0.000028] CPU: All CPU(s) started in HYP mode.
[  +0.000025] CPU: Virtualization extensions available.
[  +0.001089] devtmpfs: initialized
[  +0.016040] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[  +0.000296] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[  +0.000058] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[  +0.003338] pinctrl core: initialized pinctrl subsystem
[  +0.001156] NET: Registered protocol family 16
[  +0.004197] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[  +0.004838] audit: initializing netlink subsys (disabled)
[  +0.000296] audit: type=2000 audit(0.040:1): state=initialized audit_enabled=0 res=1
[  +0.001362] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[  +0.000039] hw-breakpoint: maximum watchpoint size is 8 bytes.
[  +0.000231] Serial: AMBA PL011 UART driver
[  +0.001966] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[  +0.025734] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-11-25 17:13, variant start_x
[  +0.010015] raspberrypi-firmware soc:firmware: Firmware hash is 2e6292a5304b95ecfd700fc5ab2b793402056332
[  +0.051830] bcm2835-dma 3f007000.dma: DMA legacy API manager, dmachans=0x1
[  +0.002174] SCSI subsystem initialized
[... 221 lines cut ...]
[  +0.237789] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[  +0.123765] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  +0.449051] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[... cut ...]
[Dec27 11:09] usb 1-1.5: USB disconnect, device number 5
[Dec27 19:10] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[  +0.133837] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[  +0.000018] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000015] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[  +0.000013] usb 1-1.5: Manufacturer: STMicroelectronics
[  +0.000013] usb 1-1.5: SerialNumber: 2040315E5034
[  +0.001192] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device

paukstelis, 2021-12-27:
I see a printer (with a serial number) and I see a camera, but for some reason things are rather jumbled up.
Please do the following:
ssh into the Pi. Plug in one camera. Give the command `dmesg` and then paste just the last 15-20 lines of the output. Then do the same for the next camera. Then do the same for each printer (individually).

enishi1357, 2021-12-27:
I plugged in each on different ports

Logitech C310 plugged in only

[   13.806138] Bluetooth: Core ver 2.22
[   13.806207] NET: Registered protocol family 31
[   13.806212] Bluetooth: HCI device and connection manager initialized
[   13.806231] Bluetooth: HCI socket layer initialized
[   13.806241] Bluetooth: L2CAP socket layer initialized
[   13.806257] Bluetooth: SCO socket layer initialized
[   13.818430] Bluetooth: HCI UART driver ver 2.3
[   13.818442] Bluetooth: HCI UART protocol H4 registered
[   13.818487] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.818647] Bluetooth: HCI UART protocol Broadcom registered
[   14.071606] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.071621] Bluetooth: BNEP filters: protocol multicast
[   14.071642] Bluetooth: BNEP socket layer initialized
[  998.756641] usb 1-1.5: USB disconnect, device number 5
[29883.020839] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[29883.154676] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, b                                                                                                                                                             cdDevice= 0.00
[29883.154694] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber                                                                                                                                                             =3
[29883.154709] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[29883.154722] usb 1-1.5: Manufacturer: STMicroelectronics
[29883.154735] usb 1-1.5: SerialNumber: 2040315E5034
[29883.155927] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[38851.178138] usb 1-1.5: USB disconnect, device number 6
[38851.178424] cdc_acm 1-1.5:1.0: failed to set dtr/rts

Ender3 skr mini e3 v3.0 board Plugged in only

[  998.756641] usb 1-1.5: USB disconnect, device number 5
[29883.020839] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[29883.154676] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[29883.154694] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[29883.154709] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[29883.154722] usb 1-1.5: Manufacturer: STMicroelectronics
[29883.154735] usb 1-1.5: SerialNumber: 2040315E5034
[29883.155927] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[38851.178138] usb 1-1.5: USB disconnect, device number 6
[38851.178424] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[39223.914766] usb 1-1.4: USB disconnect, device number 4
[39379.093360] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[39379.226965] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[39379.226983] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[39379.226997] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[39379.227011] usb 1-1.5: Manufacturer: STMicroelectronics
[39379.227024] usb 1-1.5: SerialNumber: 2040315E5034
[39379.228201] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device

PS3 Eye Plugged in only

[  998.756641] usb 1-1.5: USB disconnect, device number 5
[29883.020839] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[29883.154676] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[29883.154694] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[29883.154709] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[29883.154722] usb 1-1.5: Manufacturer: STMicroelectronics
[29883.154735] usb 1-1.5: SerialNumber: 2040315E5034
[29883.155927] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[38851.178138] usb 1-1.5: USB disconnect, device number 6
[38851.178424] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[39223.914766] usb 1-1.4: USB disconnect, device number 4
[39379.093360] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[39379.226965] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[39379.226983] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[39379.226997] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[39379.227011] usb 1-1.5: Manufacturer: STMicroelectronics
[39379.227024] usb 1-1.5: SerialNumber: 2040315E5034
[39379.228201] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[39491.694759] usb 1-1.5: USB disconnect, device number 7
[... 7 lines cut ...]
[29883.154709] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[29883.154722] usb 1-1.5: Manufacturer: STMicroelectronics
[29883.154735] usb 1-1.5: SerialNumber: 2040315E5034
[29883.155927] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[38851.178138] usb 1-1.5: USB disconnect, device number 6
[38851.178424] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[39223.914766] usb 1-1.4: USB disconnect, device number 4
[39379.093360] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[39379.226965] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[39379.226983] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[39379.226997] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[39379.227011] usb 1-1.5: Manufacturer: STMicroelectronics
[39379.227024] usb 1-1.5: SerialNumber: 2040315E5034
[39379.228201] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[39491.694759] usb 1-1.5: USB disconnect, device number 7
[39631.765561] usb 1-1.3: new full-speed USB device number 8 using dwc_otg
[... cut ...]
[39631.898855] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[39631.898869] usb 1-1.3: Product: USB Serial
[39631.975990] usbcore: registered new interface driver usbserial_generic
[39631.976045] usbserial: USB Serial support registered for generic
[39631.979012] usbcore: registered new interface driver ch341
[39631.979085] usbserial: USB Serial support registered for ch341-uart
[39631.979189] ch341 1-1.3:1.0: ch341-uart converter detected
[39631.981532] usb 1-1.3: ch341-uart converter now attached to ttyUSB0

paukstelis, 2021-12-27:
It does not seem like your cameras are getting detected at all. Both of the printers are being detected as expected. It is hard to say what the issue is in this case.
One thing to try, with both cameras plugged in give the command `ls -la /dev/video*` to see if the devices are there.

paukstelis, 2021-12-27:
And just to be clear, you are giving the dmesg command right after you plug in the device (that's what it looks like for the Anet, but I want to make sure for the cameras as well)?

enishi1357, 2021-12-27:
d

> And just to be clear, you are giving the dmesg command right after you plug in the device (that's what it looks like for the Anet, but I want to make sure for the cameras as well)?

Yes I used the dmseg command for all four.

enishi1357, 2021-12-27:
I plugged in both cameras to the ports used by the printers in case the ports were broken

crw-rw---- 1 root video 81, 7 Dec 27 22:20 /dev/video0
crw-rw---- 1 root video 81, 8 Dec 27 22:20 /dev/video1
crw-rw---- 1 root video 81, 0 Dec 27 10:52 /dev/video10
crw-rw---- 1 root video 81, 4 Dec 27 10:52 /dev/video11
crw-rw---- 1 root video 81, 6 Dec 27 10:52 /dev/video12
crw-rw---- 1 root video 81, 1 Dec 27 10:52 /dev/video13
crw-rw---- 1 root video 81, 2 Dec 27 10:52 /dev/video14
crw-rw---- 1 root video 81, 3 Dec 27 10:52 /dev/video15
crw-rw---- 1 root video 81, 5 Dec 27 10:52 /dev/video16

enishi1357, 2021-12-27:
Here is the ps3 eye on anet a8 port only

[  998.756641] usb 1-1.5: USB disconnect, device number 5
[29883.020839] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[29883.154676] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[29883.154694] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[29883.154709] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[29883.154722] usb 1-1.5: Manufacturer: STMicroelectronics
[29883.154735] usb 1-1.5: SerialNumber: 2040315E5034
[29883.155927] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[38851.178138] usb 1-1.5: USB disconnect, device number 6
[38851.178424] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[39223.914766] usb 1-1.4: USB disconnect, device number 4
[39379.093360] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[39379.226965] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[39379.226983] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[39379.226997] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[39379.227011] usb 1-1.5: Manufacturer: STMicroelectronics
[39379.227024] usb 1-1.5: SerialNumber: 2040315E5034
[39379.228201] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[39491.694759] usb 1-1.5: USB disconnect, device number 7
[39631.765561] usb 1-1.3: new full-speed USB device number 8 using dwc_otg
[39631.898838] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[39631.898855] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[39631.898869] usb 1-1.3: Product: USB Serial
[39631.975990] usbcore: registered new interface driver usbserial_generic
[39631.976045] usbserial: USB Serial support registered for generic
[39631.979012] usbcore: registered new interface driver ch341
[39631.979085] usbserial: USB Serial support registered for ch341-uart
[39631.979189] ch341 1-1.3:1.0: ch341-uart converter detected
[39631.981532] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
[41185.906803] usb 1-1.3: USB disconnect, device number 8
[41185.907763] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[41185.907897] ch341 1-1.3:1.0: device disconnected
[41196.446048] usb 1-1.4: new high-speed USB device number 9 using dwc_otg
[41196.788464] usb 1-1.4: New USB device found, idVendor=046d, idProduct=081b, bcdDevice= 0.10
[41196.788482] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=2
[41196.788496] usb 1-1.4: SerialNumber: A8EA2FC0
[41196.789531] uvcvideo: Found UVC 1.00 device <unnamed> (046d:081b)
[41196.910644] input: UVC Camera (046d:081b) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/input/input1
[41198.074718] usb 1-1.4: set resolution quirk: cval->res = 384
[41241.462781] usb 1-1.4: USB disconnect, device number 9
[41250.716213] usb 1-1.5: new high-speed USB device number 10 using dwc_otg
[41251.058591] usb 1-1.5: New USB device found, idVendor=046d, idProduct=081b, bcdDevice= 0.10
[41251.058609] usb 1-1.5: New USB device strings: Mfr=0, Product=0, SerialNumber=2
[41251.058623] usb 1-1.5: SerialNumber: A8EA2FC0
[41251.059867] uvcvideo: Found UVC 1.00 device <unnamed> (046d:081b)
[41251.180481] input: UVC Camera (046d:081b) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input2
[41252.344363] usb 1-1.5: set resolution quirk: cval->res = 384
[41379.706794] usb 1-1.5: USB disconnect, device number 10

Here is logitech webcam on ender3 port only

[  998.756641] usb 1-1.5: USB disconnect, device number 5
[29883.020839] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[29883.154676] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[29883.154694] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[29883.154709] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[29883.154722] usb 1-1.5: Manufacturer: STMicroelectronics
[29883.154735] usb 1-1.5: SerialNumber: 2040315E5034
[29883.155927] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[38851.178138] usb 1-1.5: USB disconnect, device number 6
[38851.178424] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[39223.914766] usb 1-1.4: USB disconnect, device number 4
[39379.093360] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[39379.226965] usb 1-1.5: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[39379.226983] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[39379.226997] usb 1-1.5: Product: MARLIN_STM32G0B1RE CDC in FS Mode
[39379.227011] usb 1-1.5: Manufacturer: STMicroelectronics
[39379.227024] usb 1-1.5: SerialNumber: 2040315E5034
[... 13 lines cut ...]
[41185.907763] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[41185.907897] ch341 1-1.3:1.0: device disconnected
[41196.446048] usb 1-1.4: new high-speed USB device number 9 using dwc_otg
[41196.788464] usb 1-1.4: New USB device found, idVendor=046d, idProduct=081b, bcdDevice= 0.10
[41196.788482] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=2
[41196.788496] usb 1-1.4: SerialNumber: A8EA2FC0
[41196.789531] uvcvideo: Found UVC 1.00 device <unnamed> (046d:081b)
[41196.910644] input: UVC Camera (046d:081b) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/input/input1
[41198.074718] usb 1-1.4: set resolution quirk: cval->res = 384
[41241.462781] usb 1-1.4: USB disconnect, device number 9
[41250.716213] usb 1-1.5: new high-speed USB device number 10 using dwc_otg
[... cut ...]
[41492.647142] usb 1-1.4: new high-speed USB device number 11 using dwc_otg
[41492.989531] usb 1-1.4: New USB device found, idVendor=046d, idProduct=081b, bcdDevice= 0.10
[41492.989552] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=2
[41492.989566] usb 1-1.4: SerialNumber: A8EA2FC0
[41492.993538] uvcvideo: Found UVC 1.00 device <unnamed> (046d:081b)
[41493.114315] input: UVC Camera (046d:081b) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/input/input3
[41494.279805] usb 1-1.4: set resolution quirk: cval->res = 384

paukstelis, 2021-12-28:
The Logitech has a serial number and should work. It does not look like the PS3 cam is being detected at all.

enishi1357, 2021-12-28:
How do I go about getting the logitech to work?

paukstelis, 2021-12-28:
Are you on the octoprint discord? It might be easier to work through this in real-time.
https://discord.com/invite/yA7stPp

enishi1357, 2021-12-28:
Yes.

paukstelis, 2021-12-28:
I have sent you a private message on there. I am going to close this issue.

closed 2021-12-28 by paukstelis