paukstelis/octoprint_deploy #5: Prepare on fresh Ubuntu 20.04 failed

darranwil, 2022-02-15:
Fresh install of Ubuntu 20.04, all updates, ubuntu-desktop installed.
Cloned script and ran.
Selected Prepare.
Multiple fails.
Attemped to add new instance. "Executable path is not valid! Aborting"

Removed octoprint_deploy folder, log, and .octoprint folder.
Recloned script. Ran
No option to Prepare System.

How can remove previous install and attempt again?

paukstelis, 2022-02-15:
I tested on a 20.04 VM and it worked fine, so any more info you can give from the terminal would be useful.

If you do a `sudo rm /etc/octoprint_instances` you can run Prepare System again.

darranwil, 2022-02-16:
Building wheels for collected packages: watchdog, sarge, wrapt, future, emoji, pylru, blinker, pathtools, sgmllib3k
  Building wheel for watchdog (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/darran/OctoPrint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qcze56rt/watchdog/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qcze56rt/watchdog/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-v4o28dih
       cwd: /tmp/pip-install-qcze56rt/watchdog/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for watchdog
  Running setup.py clean for watchdog
  Building wheel for sarge (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/darran/OctoPrint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qcze56rt/sarge/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qcze56rt/sarge/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-hg222ov8
       cwd: /tmp/pip-install-qcze56rt/sarge/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for sarge
  Running setup.py clean for sarge
  Building wheel for wrapt (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/darran/OctoPrint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qcze56rt/wrapt/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qcze56rt/wrapt/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-socaxzi9
       cwd: /tmp/pip-install-qcze56rt/wrapt/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for wrapt
  Running setup.py clean for wrapt
  Building wheel for future (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/darran/OctoPrint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qcze56rt/future/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qcze56rt/future/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-4cb9qet1
       cwd: /tmp/pip-install-qcze56rt/future/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for future
  Running setup.py clean for future
  Building wheel for emoji (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/darran/OctoPrint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qcze56rt/emoji/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qcze56rt/emoji/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-4azejt58
       cwd: /tmp/pip-install-qcze56rt/emoji/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for emoji
[... 36 lines cut ...]
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for pathtools
  Running setup.py clean for pathtools
  Building wheel for sgmllib3k (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/darran/OctoPrint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qcze56rt/sgmllib3k/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qcze56rt/sgmllib3k/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'" [... cut ...]
Installing collected packages: pathtools, watchdog, six, websocket-client, charset-normalizer, urllib3, certifi, idna, requests, webassets, werkzeug, markupsafe, Jinja2, Click, itsdangerous, flask, Flask-Assets, OctoPrint-PiSupport, netifaces, sgmllib3k, feedparser, sarge, pkginfo, pytz, Babel, Flask-Babel, immutabledict, cachelib, wrapt, future, regex, tornado, OctoPrint-FirmwareCheck, PyYAML, emoji, markdown, netaddr, pyserial, ifaddr, zeroconf, semantic-version, Flask-Login, psutil, sentry-sdk, pylru, unidecode, zipstream-new, OctoPrint-FileCheck, pathvalidate, filetype, blinker, colorlog, OctoPrint

paukstelis, 2022-02-16:
I also saw some of these errors, but it had no impact on the installation. Do you have /home/$USER/OctoPrint present?

darranwil, 2022-02-16:
I do this time! Yay! Only issue for first instance was service wasn't enabled but that is easy enough. 
Webcam service isnt starting. I do see in the cam_generic.serivce the paths are missing a "e" in the directory name.
Corrected that and enabled service. Fails with:

Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: starting application
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]:  i: Using V4L2 device.: /dev/video0
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]:  i: Desired Resolution: 1920 x 1080
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]:  i: Frames Per Second.: 30
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]:  i: Format............: JPEG
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]:  i: TV-Norm...........: DEFAULT
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: ERROR opening V4L interface: Permission denied
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]:  i: init_VideoIn failed
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: Using V4L2 device.: /dev/video0
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: Desired Resolution: 1920 x 1080
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: Frames Per Second.: 30
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: Format............: JPEG
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: TV-Norm...........: DEFAULT
Feb 16 03:25:33 miniserver1 mjpg_streamer[4788]: MJPG-streamer [4788]: init_VideoIn failed
Feb 16 03:25:33 miniserver1 sudo[4785]: pam_unix(sudo:session): session closed for user root
Feb 16 03:25:33 miniserver1 systemd[1]: cam_generic.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited

darranwil, 2022-02-16:
manually install v4l-utils and got past that error. Now 2 more, lol.

     Loaded: loaded (/etc/systemd/system/cam_generic.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-02-16 03:41:50 UTC; 6min ago
    Process: 893 ExecStart=/home/darran/mjpeg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_generic -r 1920x1080 -f 30 -o output_http.so -p 8001 (code=exited, status=1/FAILURE)
   Main PID: 893 (code=exited, status=1/FAILURE)

Feb 16 03:41:50 miniserver1 mjpg_streamer[893]:  i: init_VideoIn failed
Feb 16 03:41:50 miniserver1 systemd[1]: cam_generic.service: Failed with result 'exit-code'.
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: starting application
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: Using V4L2 device.: /dev/video0
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: Desired Resolution: 1920 x 1080
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: Frames Per Second.: 30
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: Format............: JPEG
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: TV-Norm...........: DEFAULT
Feb 16 03:41:50 miniserver1 unknown[893]: MJPG-streamer [893]: init_VideoIn failed

v4l2-ctl --list-devices gives me:
Video Capture 3 (usb-0000:00:15.0-2):
        /dev/video0
        /dev/video1
        /dev/media0

Also Octoprint will not connect to printer. SKR1.4 Turbo running Marlin 2.0. 
Tried manually connecting with /dev/ttyACM0 and using Serial Port: AUTO with known working baud rate and AUTO

dmesg | grep tty points to ttyACM0

paukstelis, 2022-02-16:
you should not have to connect to the printer with AUTO. Once you set up an instance with the script it will be available as /dev/octo_instancename. It may be that you need to log out/log in in order to connect. The user gets added to dialout group, but that change probably does not go into effect until after a logout/reboot.

I can't do much for the cameras. It says 'experimental' for a reason! I will add  v4l-utils to the package list.

darranwil, 2022-02-16:
Yeah I rebooted and manually added my account to the dialout and tty group after several reboots and it not connecting just to make sure.  No worries, just giving feedback since this is a great script you've put together. Any resolutions I find I will add to this thread.

darranwil, 2022-02-16:
Completely confused, lol.  Shut down both the pc and printer multiple times yesterday and today OctoPrint connects! No clue why.

For the webcam I had to do:  "sudo chmod 777 /dev/video*"

That made the webcam available to Cheese in desktop, with only 640x320 resolution available, but progress.  

Cam_generic.service now starts up but www-folder-path shows disabled and browser returns a 501 error.

Added "-w /home/darran/mjpeg-streamer/www/" to the cam_generic.serivce and now mjpeg-streamer is fully up.

darranwil, 2022-02-27:
> > Added "-w /home/darran/mjpeg-streamer/www/" to the cam_generic.serivce and now mjpeg-streamer is fully up.
> 
> where did you add this line?

Added to the cam_generic.service file under /etc/systemd/system/ 

Anytime you edit a .service file you will have to run a "sudo systemctl daemon-reload" to make the changes active.

BUT, after adding 2 instances for my 2 printers and cameras it went away and are replaced with cam_instancename.service.  I didn't have to add it to them except I did have to correct the Environment= and ExecStart= paths.  Both had mjpg-streamer as the folder name instead of mjpeg-streamer for the folder names. The mjpg_streamer was correct.  I also had to fix the path in both as they had an extra "/" in the paths.  I then had to add the username of the user octoprint was installed and running under to the video group with "sudo usermod -a -G video exampleusername".

The previous step I tried of "sudo chmod 777 /dev/video" only lasted till reboot and adding the user to the video group resolved that across reboots.

Here is what my cam_printer1.service looks like currently:
-----------------------------
[Unit]
Description=the OctoPi(buntu) webcam daemon with the user specified config
After=network.online.target
Wants=network.online.target

[Service]
User=darran
Environment="LD_LIBRARY_PATH=/home/darran/mjpeg-streamer"
ExecStart=/home/darran/mjpeg-streamer/mjpg_streamer -i "input_uvc.so -d /dev/cam_tronxyxxx3 -r 1920x1080 -f 30" -o "output_http.so >

[Install]
WantedBy=multi-user.target
-----------------------------

paukstelis, 2022-02-27:
There was an error in the Ubuntu prepare script that gave mjpg-streamer the
wrong path in the service file. It has since been fixed and you shouldn't
have to modify the file.

closed 2022-06-09 by paukstelis