paukstelis/octoprint_deploy #8: Trouble setting up multi-instances

theohenrichs, 2022-03-11:
Hi Paul

I am stuck setting-up a multi instance Octoprint with your script. It wil not run as aspected.
This is what I encounter:

I did put the latest OctoPi image on my SD card
Then via PuTTY I connected the Pi with pi@octopi.local and got the script: 
This is what I see on the screen:

pi@octopi:~ $ git clone https://github.com/paukstelis/octoprint_deploy.git

I choose Prepare System from the menu

Then I did Setup the admin user by connecting to http://octopi.local via browser

Then back in the sesion I choose 1) New Instance ->>

pi@octopi:~ $ sudo octoprint_deploy/octoprint_deploy.sh
1) New instance
2) Delete instance
3) Add Camera
4) USB port testing
5) Quit
Select operation: 1
Fri 11 Mar 11:58:14 CET 2022 starting instance installation
1) OctoPi
2) OctoBuntu
3) Other
4) Quit
Installation type: 1
UNPLUG PRINTER YOU ARE INSTALLING NOW (other printers can remain)
Enter the name for new printer/instance (no spaces):
3dgear
Port on which this instance will run (ENTER will increment from last value in /etc/octoprint_instances):
Selected port is: 5001
Octoprint Daemon User [pi]:
Octoprint Executable Daemon Path [/home/pi/oprint/bin/octoprint]:
Executable path is valid
Octoprint Config Path (where the hidden instance directory will be) [/home/pi/]:
Octoprint instance template path [/home/pi/.octoprint]:
Template path is valid
**WARNING!! You must setup the base profile and admin user before continuing**

Can you tell me what is wrong?

paukstelis, 2022-03-11:
>    6. OctoPi
>    7. OctoBuntu
>    8. Other
>    9. Quit

This means you haven't completed the first run setup. You have to setup the
admin user and all the other steps of the first run before making a new
instance.

theohenrichs, 2022-03-11:
Hi Paul

Thanks for the answer. I was aware of the fact that the admin user had to be setup. So I did so, and even connected the printer with this user and did some commands. (see attached screenshot)
The result is still the same:
"WARNING!! You must setup the base profile and admin user before continuing"

I stil do not know what else I have to do to show the script that I have setup the admin user.

[image]

paukstelis, 2022-03-11:
Another thing you might try is to restart the octoprint instance. This is not an issue I have ever seen before.

theohenrichs, 2022-03-11:
Hi Paul

This is the content of config.yaml
pi@octopi:~ $ cat .octoprint/config.yaml
appearance:
  name: INSTANCE
feature:
  modelSizeDetection: false
plugins:
  discovery:
    upnpUuid: 56e96976-3233-4ddd-840a-0392dcf4c9b0
serial:
  additionalPorts:
  - /dev/octo_INSTANCE
server:
  commands:
    serverRestartCommand: sudo systemctl restart INSTANCE
    systemRestartCommand: sudo reboot
  firstRun: true
temperature:
  profiles:
  - bed: 100
    chamber: null
    extruder: 210
    name: ABS
  - bed: 60
    chamber: null
    extruder: 180
    name: PLA

All looks oké regarding the firstRun parameter.
I did a reboot of the Pi and a restart of the Octoprint instance did not change the result:  "WARNING!! You must setup the base profile and admin user before continuing"

paukstelis, 2022-03-11:
`firstRun: true` will result in the script believing that the admin user has not been setup. That should get changed to false after the first run is complete. Why it has not in this case, I have no idea. When you reboot the Pi does it ask you to setup the admin user again?

theohenrichs, 2022-03-11:
No, when I reboot, the Pi does not ask me to setup the admin user again

paukstelis, 2022-03-11:
I'm stumped in that case. If you have firstRun: true in config.yaml it should ask you for that. I'm assuming you are using an OctoPi image and not installing octoprint manually on Raspian?

paukstelis, 2022-03-11:
You can try just editing config.yaml and switch firstRun to false.

theohenrichs, 2022-03-11:
I Just editted the config.yaml and switched firstRun into false.
The process did continue now (wow) and I was able to make 2 extra instances.
So far so good . . . .
I have to put in de portnbr after the http://octopi.local:5001. It is translated to 
http://octopi.local:5001/login/?redirect=%2F%3F&permissions=STATUS%2CSETTINGS_READ
and now the admin credentials are not working. Can I correct that?
[image]

theohenrichs, 2022-03-11:
Perhaps is was better to use the URL:  http://octopi.local/3dgear   but than the following appears
[image]
 so I checked:
pi@octopi:~ $ ps -ef | grep -i octoprint | grep -i python
pi        6117     1  1 20:48 ?        00:00:22 /home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serve --config=/home/pi//.3dgear/config.yaml --basedir=/home/pi//.3dgear --port=5001
pi        7230     1  1 20:54 ?        00:00:19 /home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serve --config=/home/pi//.mendel90/config.yaml --basedir=/home/pi//.mendel90 --port=5002

Although I do not understand all of it, but it seems oké, so why are the admin-credentials invalid

paukstelis, 2022-03-12:
I feel like there is something wrong from the start. Do you have a users.yaml in /home/pi/.octoprint and in /home/pi/.3dgear ?

theohenrichs, 2022-03-12:
Hi Paul I much appreciate your help!
Yes I have!

pi@octopi:~/.octoprint $ cd /home/pi/.octoprint
pi@octopi:~/.octoprint $ ls
config.backup  data       logs     printerProfiles  slicingProfiles  translations  users.yaml  watched
config.yaml    generated  plugins  scripts          timelapse        uploads       virtualSd
pi@octopi:~/.octoprint $ cd /home/pi/.3dgear
pi@octopi:~/.3dgear $ ls
config.backup  data       logs     printerProfiles  slicingProfiles  translations  users.yaml  watched
config.yaml    generated  plugins  scripts          timelapse        uploads       virtualSd
pi@octopi:~/.3dgear $

and both files users.yaml are identical.

P.S.
Perhaps I can try again from the start with a new image from scratch?

theohenrichs, 2022-03-12:
Hi Paul

As I stated in the former comment I started again from scratch.
I managed to make a clean instal from scratch on a 32GB SD-card (in stead of the 64GB SD) there were no problems during the installation nor during making the 2 instances. 

The only thing I see is that the initial URL http://octopi.local to setup the admin account did work, but the URL http://octopi.local/3dgear did NOT work. I overcame this by typing the IP-address with portnbr.

I am now satisfied with the result and very happy with you help. I buy you a coffee!!

paukstelis, 2022-03-12:
I'm glad to hear this! It did seem that something had gone wrong early on.

Usually when octopi.local cannot be reached it is related to the router, however if you could reach it for the initial setup, it should also work for the other instances. You could check the contents of /etc/haproxy/haproxy.cfg and make sure the haproxy services is running: `sudo systemctl status haproxy.service`

closed 2022-06-09 by paukstelis