paukstelis/octoprint_deploy #10: Second cam on second instance failing to start

pebo48, 2022-04-03:
Hi Paul I`m having trouble with two instances and two cameras, I have both instances up and running and a camera working on first instance but can not get a camera to work on second instance both cams are C270 webcams. I have included both status cam instances below and it look like one has failed, I`m new to all this and am not sure what is going on

● cam_CR10s.service - the OctoPi(buntu) webcam daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/cam_CR10s.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2022-04-03 10:20:14 BST; 1h 44min ago
 Main PID: 395 (mjpg_streamer)
    Tasks: 4 (limit: 3596)
   CGroup: /system.slice/cam_CR10s.service
           └─395 /home/pi/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_CR10s -r 640x480 -f 5 -o output_http.so -p 8001

Apr 03 10:20:14 octopi mjpg_streamer[395]:  o: HTTP Listen Address..: (null)
Apr 03 10:20:14 octopi mjpg_streamer[395]:  o: username:password....: disabled
Apr 03 10:20:14 octopi mjpg_streamer[395]:  o: commands.............: enabled
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: www-folder-path......: disabled
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: HTTP TCP port........: 8001
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: HTTP Listen Address..: (null)
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: username:password....: disabled
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: commands.............: enabled
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: starting input plugin input_uvc.so
Apr 03 10:20:14 octopi mjpg_streamer[395]: MJPG-streamer [395]: starting output plugin: output_http.so (ID: 00)

● cam_Ender3.service - the OctoPi(buntu) webcam daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/cam_Ender3.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-04-03 10:20:14 BST; 1h 34min ago
  Process: 400 ExecStart=/home/pi/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_Ender3 -r 640x480 -f 5 -o output_http.so -p 8002 (code=exited, status=1/FAILURE)
 Main PID: 400 (code=exited, status=1/FAILURE)

Apr 03 10:20:14 octopi mjpg_streamer[400]: MJPG-streamer [400]: Frames Per Second.: 5
Apr 03 10:20:14 octopi mjpg_streamer[400]: MJPG-streamer [400]: Format............: JPEG
Apr 03 10:20:14 octopi mjpg_streamer[400]: MJPG-streamer [400]: TV-Norm...........: DEFAULT
Apr 03 10:20:14 octopi mjpg_streamer[400]: MJPG-streamer [400]: init_VideoIn failed
Apr 03 10:20:14 octopi mjpg_streamer[400]: libv4l2: error setting pixformat: Device or resource busy
Apr 03 10:20:14 octopi mjpg_streamer[400]: Unable to set format: 1196444237 res: 640x480
Apr 03 10:20:14 octopi mjpg_streamer[400]: Init v4L2 failed !! exit fatal
Apr 03 10:20:14 octopi mjpg_streamer[400]:  i: init_VideoIn failed
Apr 03 10:20:14 octopi systemd[1]: cam_Ender3.service: Main process exited, code=exited, status=1/FAILURE
Apr 03 10:20:14 octopi systemd[1]: cam_Ender3.service: Failed with result 'exit-code'.

thank you for all the help

paukstelis, 2022-04-03:
Let's try two things first (this is all in an ssh terminal to the pi; paste the results):
`cat /etc/udev/rules.d/99-octoprint.rules`
and
`ls -la /dev/cam*`

pebo48, 2022-04-03:
pi@octopi:~ $ cat /etc/udev/rules.d/99-octoprint.rules
SUBSYSTEM=="tty", ATTRS{serial}=="AB0LZ1EY", SYMLINK+="octo_CR10s"
SUBSYSTEM=="video4linux", ATTRS{serial}=="200901010001", ATTR{index}=="0", SYMLINK+="cam_CR10s"
KERNELS=="1-1.2.4:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3"
SUBSYSTEM=="video4linux", ATTRS{serial}=="200901010001", ATTR{index}=="0", SYMLINK+="cam_Ender3"

pebo48, 2022-04-03:
pi@octopi:~ $ ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  3 10:20 /dev/cam_CR10s -> video0
lrwxrwxrwx 1 root root 6 Apr  3 10:20 /dev/cam_Ender3 -> video0

pebo48, 2022-04-03:
both cams seem to have same serial number and on the boxes they are two different numbers also the serial number showing does not match any serial on either camera box

paukstelis, 2022-04-03:
OK. The problem is clear, but not why it happened. The serial numbers are the same for both cameras. That shouldn't happen with a Logitech camera (shouldn't, doesn't mean it doesn't). 
We need to get a few pieces of information and it will be possible to edit so it will work (I think).

First, unplug and replug the ender camera, then in the ssh session give the command `dmesg`
Paste about the last 10 or 15 lines from the output and we will determine which USB port it is plugged into to (it will have to stay in that position).

pebo48, 2022-04-03:
[11863.738435] usb 1-1.2.4: USB disconnect, device number 12
[11863.738930] usb 1-1.2.4: failed to send control message: -19
[11863.739220] ch341-uart ttyUSB0: ch341-uart converter now disconnected from tt                          yUSB0
[11863.739312] ch341 1-1.2.4:1.0: device disconnected
[14995.068869] usb 1-1.2.3: USB disconnect, device number 5
[15002.041990] usb 1-1.2.3: new high-speed USB device number 13 using xhci_hcd
[15002.286777] usb 1-1.2.3: New USB device found, idVendor=046d, idProduct=0825,                           bcdDevice= 0.21
[15002.286797] usb 1-1.2.3: New USB device strings: Mfr=0, Product=1, SerialNumb                          er=2
[15002.286813] usb 1-1.2.3: Product: C270 HD WEBCAM
[15002.286828] usb 1-1.2.3: SerialNumber: 200901010001
[15002.298635] uvcvideo: Found UVC 1.00 device C270 HD WEBCAM (046d:0825)
[15002.304547] input: C270 HD WEBCAM as /devices/platform/scb/fd500000.pcie/pci0                          000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.3/1-1.2.3:1.0/input/input2
[15002.310967] usb 1-1.2.3: current rate 16000 is different from the runtime rat                          e 32000
[15002.316379] usb 1-1.2.3: current rate 24000 is different from the runtime rat                          e 16000
[15002.319487] usb 1-1.2.3: 3:3: cannot set freq 24000 to ep 0x82
[15002.446442] usb 1-1.2.3: set resolution quirk: cval->res = 384

paukstelis, 2022-04-03:
OK. First let's delete that Ender3 camera entry:
`sudo sed -i "/cam_Ender3/d" /etc/udev/rules.d/99-octoprint.rules`

Then update based on USB port position (command should all be on one line!):
`sudo echo SUBSYSTEM==\"video4linux\",KERNELS==\"1-1.2.3:1.0\",SUBSYSTEMS==\"usb\",ATTR{index}==\"0\",DRIVERS==\"uvcvideo\",SYMLINK+=\"cam_Ender3\" >> /etc/udev/rules.d/99-octoprint.rules`

Make sure nothing is printing on any instance, then do 
`sudo udevadm control --reload-rules`
`sudo  udevadm trigger`

paukstelis, 2022-04-03:
Then do these two commands again.

> Let's try two things first (this is all in an ssh terminal to the pi; paste the results): `cat /etc/udev/rules.d/99-octoprint.rules` and `ls -la /dev/cam*`

pebo48, 2022-04-03:
pi@octopi:~ $ sudo sed -i "/cam_Ender3/d" /etc/udev/rules.d/99-octoprint.rules
[sudo] password for pi:
pi@octopi:~ $ sudo echo SUBSYSTEM==\"video4linux\",KERNELS==\"1-1.2.3:1.0\",SUBSYSTEMS==\"usb\",ATTR{index}==\"0\",DRIVERS==\"uvcvideo\",SYMLINK+=\"cam_Ender3\" >> /etc/udev/rules.d/99-octoprint.rules
-bash: /etc/udev/rules.d/99-octoprint.rules: Permission denied
pi@octopi:~ $ sudo udevadm control --reload-rules
pi@octopi:~ $ sudo udevadm trigger
pi@octopi:~ $ cat /etc/udev/rules.d/99-octoprint.rules
SUBSYSTEM=="tty", ATTRS{serial}=="AB0LZ1EY", SYMLINK+="octo_CR10s"
SUBSYSTEM=="video4linux", ATTRS{serial}=="200901010001", ATTR{index}=="0", SYMLINK+="cam_CR10s"
KERNELS=="1-1.2.4:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3"
pi@octopi:~ $ ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  3 14:48 /dev/cam_CR10s -> video2

paukstelis, 2022-04-03:
Something weird with echoing commands with quotes in them I think prevented it from working. One moment, let me figure it out.

paukstelis, 2022-04-03:
This should do it:
`echo SUBSYSTEM==\"video4linux\",KERNELS==\"1-1.2.3:1.0\",SUBSYSTEMS==\"usb\",ATTR{index}==\"0\",DRIVERS==\"uvcvideo\",SYMLINK+=\"cam_Ender3\" | sudo tee -a /etc/udev/rules.d/99-octoprint.rules`

then try those series of commands again (udevadm, etc.)

paukstelis, 2022-04-03:
This does bring up that I should probably do serial number checks during the detection process to make sure there are no duplicates!

pebo48, 2022-04-03:
pi@octopi:~ $ cat /etc/udev/rules.d/99-octoprint.rules
SUBSYSTEM=="tty", ATTRS{serial}=="AB0LZ1EY", SYMLINK+="octo_CR10s"
SUBSYSTEM=="video4linux", ATTRS{serial}=="200901010001", ATTR{index}=="0", SYMLINK+="cam_CR10s"
KERNELS=="1-1.2.4:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3"
SUBSYSTEM=="video4linux",KERNELS=="1-1.2.3:1.0",SUBSYSTEMS=="usb",ATTR{index}=="0",DRIVERS=="uvcvideo",SYMLINK+="cam_Ender3"
pi@octopi:~ $ ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  3 14:59 /dev/cam_CR10s -> video2
lrwxrwxrwx 1 root root 6 Apr  3 14:59 /dev/cam_Ender3 -> video2
pi@octopi:~ $

pebo48, 2022-04-03:
do i need to reboot the pi

paukstelis, 2022-04-03:
Well, that's no good. Still pointing to the same camera. What you may have to do in this case is have them both based on usb position. Run through the same instructions with `dmesg` with CR10 camera and I'll give you commands to modify that one as well.

pebo48, 2022-04-03:
17136.275851] usb 1-1.2.1: new high-speed USB device number 15 using xhci_hcd
[17136.520523] usb 1-1.2.1: New USB device found, idVendor=046d, idProduct=0825, bcdDevice= 0.21
[17136.520541] usb 1-1.2.1: New USB device strings: Mfr=0, Product=1, SerialNumber=2
[17136.520554] usb 1-1.2.1: Product: C270 HD WEBCAM
[17136.520568] usb 1-1.2.1: SerialNumber: 200901010001
[17136.531534] uvcvideo: Found UVC 1.00 device C270 HD WEBCAM (046d:0825)
[17136.537411] input: C270 HD WEBCAM as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.1/1-1.2.1:1.0/input/input3
[17136.543770] usb 1-1.2.1: current rate 16000 is different from the runtime rate 32000
[17136.549863] usb 1-1.2.1: current rate 24000 is different from the runtime rate 16000
[17136.552969] usb 1-1.2.1: 3:3: cannot set freq 24000 to ep 0x82
[17136.677999] usb 1-1.2.1: set resolution quirk: cval->res = 384
pi@octopi:~ $

paukstelis, 2022-04-03:
`sudo sed -i "/cam_CR10s/d" /etc/udev/rules.d/99-octoprint.rules`
and
`echo SUBSYSTEM==\"video4linux\",KERNELS==\"1-1.2.1:1.0\",SUBSYSTEMS==\"usb\",ATTR{index}==\"0\",DRIVERS==\"uvcvideo\",SYMLINK+=\"cam_CR10s\" | sudo tee -a /etc/udev/rules.d/99-octoprint.rules`
followed with the udevadm commands

pebo48, 2022-04-03:
i@octopi:~ $ sudo sed -i "/cam_CR10s/d" /etc/udev/rules.d/99-octoprint.rules
pi@octopi:~ $ echo SUBSYSTEM==\"video4linux\",KERNELS==\"1-1.2.1:1.0\",SUBSYSTEMS==\"usb\",ATTR{index}==\"0\",DRIVERS==\"uvcvideo\",SYMLINK+=\"cam_CR10s\" | sudo tee -a /etc/udev/rules.d/99-octoprint.rules
SUBSYSTEM=="video4linux",KERNELS=="1-1.2.1:1.0",SUBSYSTEMS=="usb",ATTR{index}=="0",DRIVERS=="uvcvideo",SYMLINK+="cam_CR10s"
pi@octopi:~ $ sudo udevadm control --reload-rules
pi@octopi:~ $ sudo udevadm trigger
pi@octopi:~ $ cat /etc/udev/rules.d/99-octoprint.rules
SUBSYSTEM=="tty", ATTRS{serial}=="AB0LZ1EY", SYMLINK+="octo_CR10s"
KERNELS=="1-1.2.4:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3"
SUBSYSTEM=="video4linux",KERNELS=="1-1.2.3:1.0",SUBSYSTEMS=="usb",ATTR{index}=="0",DRIVERS=="uvcvideo",SYMLINK+="cam_Ender3"
SUBSYSTEM=="video4linux",KERNELS=="1-1.2.1:1.0",SUBSYSTEMS=="usb",ATTR{index}=="0",DRIVERS=="uvcvideo",SYMLINK+="cam_CR10s"
pi@octopi:~ $ ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  3 15:11 /dev/cam_CR10s -> video1
lrwxrwxrwx 1 root root 6 Apr  3 15:11 /dev/cam_Ender3 -> video2
pi@octopi:~ $

pebo48, 2022-04-03:
no cameras are showing do i need to restart each instance

paukstelis, 2022-04-03:
Yes, you need to restart the camera instances:
sudo systemctl restart cam_CR10s
sudo systemctl restart cam_Ender3

pebo48, 2022-04-03:
ok Ender3 cam showing but not CR10s

paukstelis, 2022-04-03:
`sudo systemctl status cam_CR10s` will likely say why

A reboot might not be a bad idea either.

pebo48, 2022-04-03:
just did and both cameras now showing on respective instances.......Thank you so much I will have to buy you a coffee

pebo48, 2022-04-03:
so just to confirm both cams have to stay in their respective ports no moving the am I correct

paukstelis, 2022-04-03:
That is correct. They have to be in those spots to work.

closed 2022-04-03 by pebo48