paukstelis/octoprint_deploy #11: Error when preparing system, big noob here.

LehmannADL, 2022-04-05:
This only needs to be run once to prepare your system to use octoprint_deploy.
Run this setup and then connect to OctoPrint through your browser to setup your admin user.
1) OctoPi
2) Ubuntu 18-20
3) Ubuntu 21.X
4) Mint 20.3
5) Fedora 35+
6) Quit
Installation type: 1
Ready to begin? [y/n]: y
Adding camera port records
Adding current user to dialout and video groups.
Disabling unneeded services....
octoprint.service is not a native service, redirecting to systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable octoprint
Failed to disable unit: Unit file webcamd.service does not exist.
Failed to stop webcamd.service: Unit webcamd.service not loaded.
Failed to disable unit: Unit file streamer_select.service does not exist.
Failed to stop streamer_select.service: Unit streamer_select.service not loaded.
Adding systemctl and reboot to sudo
Modifying config.yaml
cp: cannot create regular file '/home/pi/.octoprint/config.yaml': No such file or directory

very lost here hope someone is willing to help.

paukstelis, 2022-04-05:
Did you actually put OctoPi on your SD card (not raspian or some other operating system)? It would appear that OctoPrint is not installed, which suggests that it is not actually OctoPi you have.

[image]

LehmannADL, 2022-04-05:
i have it running on ubuntu server 20.04.4
my octoprint server is up and running fine.

[https://ibb.co/djw17Th](https://ibb.co/djw17Th)

paukstelis, 2022-04-05:
In that case, you don't want to be selecting 1 for OctoPi!
You should be selecting 2 for Ubuntu. But since you already have installed OctoPrint, there may be some redundancy, and since I don't know what has already been done to get the server running I really can't help sort out those issues. Best thing to do is start with a fresh installation of Ubuntu and follow the instructions for Ubuntu in the README.

LehmannADL, 2022-04-05:
i followed these instructions, are they wrong?

paukstelis, 2022-04-05:
No, the instructions are right, but you have clearly selected 'OctoPi' in `Prepare System` when you are using Ubuntu. You need to choose Ubuntu when you are using Ubuntu. The setup is different for different  operating systems.

LehmannADL, 2022-04-05:
now im stuck here.....

Last login: Tue Apr  5 19:59:27 2022 from 10.0.0.14
a@a:~$ sudo octoprint_deploy/octoprint_deploy.sh
[sudo] password for a:
1) New instance      3) Add Camera        5) Quit
2) Delete instance   4) USB port testing
Select operation: 1
Tue 05 Apr 2022 08:15:23 PM UTC starting instance installation
1) OctoPi
2) OctoBuntu
3) Other
4) Quit
Installation type: 2
UNPLUG PRINTER YOU ARE INSTALLING NOW (other printers can remain)
Enter the name for new printer/instance (no spaces):
SV04
Port on which this instance will run (ENTER will increment from last value in /etc/octoprint_instances):

Selected port is: 5001
Octoprint Daemon User [a]:

Octoprint Executable Daemon Path [/home/a/OctoPrint/bin/octoprint]:

Executable path is valid
Octoprint Config Path (where the hidden instance directory will be) [/home/a/]:

Octoprint instance template path [/home/a/.octoprint]:

Template path is valid
WARNING!! You must setup the base profile and admin user before continuing
a@a:~$

LehmannADL, 2022-04-05:
now the password on my original octoprint is differant and i cant even login.....

LehmannADL, 2022-04-05:
and the second server never starts so i cant setup the base profile

paukstelis, 2022-04-05:
If you started from the beginning there should not be an original octoprint. The script literally does everything for you, including installing octoprint. This is why I suggested that you start from a clean Ubuntu installation. Just totally reinstall Ubuntu and follow the README guide.

LehmannADL, 2022-04-05:
oh so sorry i didnt know

LehmannADL, 2022-04-06:
OK AWESOME i got almost everything perfect the only thing i cant get to work is the second camera. and im assuming it might have to do with the fact that they are both the same camera? i followed your readme added everything. just cant get that second cam to stream.

paukstelis, 2022-04-06:
It may be a case where the second camera has the same serial number as the first which leads to issues.
Read though this: https://github.com/paukstelis/octoprint_deploy/issues/10