paukstelis/octoprint_deploy #12: Issue #10 Closed, still cant get that second camera to work

LehmannADL, 2022-04-06:
adam@octoprint:~$ ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  6 14:57 /dev/cam_LulzMini -> video0
lrwxrwxrwx 1 root root 6 Apr  6 14:57 /dev/cam_SV04 -> video3

adam@octoprint:~$ cat /etc/udev/rules.d/99-octoprint.rules
KERNELS=="5-1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_SV04"
SUBSYSTEM=="tty", ATTRS{serial}=="7403530343035131C061", SYMLINK+="octo_LulzMini                                                                          "
SUBSYSTEM=="video4linux",KERNELS=="2-1.6:1.0",SUBSYSTEMS=="usb",ATTR{index}=="0"                                                                          ,DRIVERS=="uvcvideo",SYMLINK+="cam_LulzMini"
SUBSYSTEM=="video4linux",KERNELS=="5-2:1.0",SUBSYSTEMS=="usb",ATTR{index}=="0",D                                                                          RIVERS=="uvcvideo",SYMLINK+="cam_SV04"

DMSEG SV04 CAMERA

[   11.351710] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[   17.148645] usb 5-2: USB disconnect, device number 3
[   17.477553] usb 5-2: new high-speed USB device number 4 using xhci_hcd
[   17.689634] usb 5-2: device descriptor read/64, error -71
[   18.005242] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[   18.005244] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[   18.005245] usb 5-2: Product: USB 2.0 Camera
[   18.005246] usb 5-2: Manufacturer: Sonix Technology Co., Ltd.
[   18.005246] usb 5-2: SerialNumber: SN0001
[   18.009674] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[   18.022473] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input19
[   18.030081] usb 5-2: 3:1: cannot get freq at ep 0x84
[  369.000782] usb 5-2: USB disconnect, device number 4
[  374.371560] usb 5-2: new high-speed USB device number 5 using xhci_hcd
[  374.627711] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[  374.627714] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  374.627716] usb 5-2: Product: USB 2.0 Camera
[  374.627717] usb 5-2: Manufacturer: Sonix Technology Co., Ltd.
[  374.627719] usb 5-2: SerialNumber: SN0001
[  374.632225] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[  374.645107] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input20
[  374.652749] usb 5-2: 3:1: cannot get freq at ep 0x84
[  376.520396] usb 5-2: USB disconnect, device number 5
[  376.847322] usb 5-2: new high-speed USB device number 6 using xhci_hcd
[  377.183391] usb 5-2: device descriptor read/64, error -71
[  377.419375] usb 5-2: device descriptor read/64, error -71
[  377.655241] usb 5-2: new high-speed USB device number 7 using xhci_hcd
[  378.199292] usb 5-2: device descriptor read/64, error -71
[  378.515263] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[  378.515267] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  378.515269] usb 5-2: Product: USB 2.0 Camera
[  378.515270] usb 5-2: Manufacturer: Sonix Technology Co., Ltd.
[  378.515272] usb 5-2: SerialNumber: SN0001
[  378.519843] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[  378.532816] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input21
[  378.540463] usb 5-2: 3:1: cannot get freq at ep 0x84

DMSEG LULZMINI CAMERA

[   17.148645] usb 5-2: USB disconnect, device number 3
[   17.477553] usb 5-2: new high-speed USB device number 4 using xhci_hcd
[   17.689634] usb 5-2: device descriptor read/64, error -71
[   18.005242] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[   18.005244] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[   18.005245] usb 5-2: Product: USB 2.0 Camera
[   18.005246] usb 5-2: Manufacturer: Sonix Technology Co., Ltd.
[   18.005246] usb 5-2: SerialNumber: SN0001
[   18.009674] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[   18.022473] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input19
[   18.030081] usb 5-2: 3:1: cannot get freq at ep 0x84
[  369.000782] usb 5-2: USB disconnect, device number 4
[  374.371560] usb 5-2: new high-speed USB device number 5 using xhci_hcd
[  374.627711] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[  374.627714] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  374.627716] usb 5-2: Product: USB 2.0 Camera
[  374.627717] usb 5-2: Manufacturer: Sonix Technology Co., Ltd.
[  374.627719] usb 5-2: SerialNumber: SN0001
[  374.632225] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[  374.645107] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input20
[  374.652749] usb 5-2: 3:1: cannot get freq at ep 0x84
[  376.520396] usb 5-2: USB disconnect, device number 5
[  376.847322] usb 5-2: new high-speed USB device number 6 using xhci_hcd
[  377.183391] usb 5-2: device descriptor read/64, error -71
[  377.419375] usb 5-2: device descriptor read/64, error -71
[  377.655241] usb 5-2: new high-speed USB device number 7 using xhci_hcd
[  378.199292] usb 5-2: device descriptor read/64, error -71
[  378.515263] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[  378.515267] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  378.515269] usb 5-2: Product: USB 2.0 Camera
[  378.515270] usb 5-2: Manufacturer: Sonix Technology Co., Ltd.
[  378.515272] usb 5-2: SerialNumber: SN0001
[  378.519843] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[  378.532816] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input21
[  378.540463] usb 5-2: 3:1: cannot get freq at ep 0x84
[  586.384892] usb 5-1: USB disconnect, device number 2
[  586.385961] usb 5-1: failed to send control message: -19
[  586.386202] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[  586.386226] ch341 5-1:1.0: device disconnected
[  590.412967] usb 5-1: new full-speed USB device number 8 using xhci_hcd
[  590.562079] usb 5-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[  590.562082] usb 5-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  590.562084] usb 5-1: Product: USB2.0-Serial
[  590.569378] ch341 5-1:1.0: ch341-uart converter detected
[  590.569876] usb 5-1: ch341-uart converter now attached to ttyUSB0

LehmannADL, 2022-04-06:
nvm i think i see where i messed up

LehmannADL, 2022-04-06:
i have no idea whats wrong. other then under

ls -la /dev/cam*
lrwxrwxrwx

LehmannADL, 2022-04-06:
adam@octoprint:~$ sudo systemctl status cam_LulzMini
● cam_LulzMini.service - the OctoPi(buntu) webcam daemon with the user specifie>
     Loaded: loaded (/etc/systemd/system/cam_LulzMini.service; enabled; vendor >
     Active: active (running) since Wed 2022-04-06 14:57:19 UTC; 33min ago
   Main PID: 782 (mjpg_streamer)
      Tasks: 4 (limit: 9079)
     Memory: 2.2M
     CGroup: /system.slice/cam_LulzMini.service
             └─782 /home/adam/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /d>

Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: Format......>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: TV-Norm.....>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: FPS coerced >
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: www-folder-p>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: HTTP TCP por>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: HTTP Listen >
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: username:pas>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: commands....>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: starting inp>
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: starting out>
lines 1-19/19 (END)
● cam_LulzMini.service - the OctoPi(buntu) webcam daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam_LulzMini.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-04-06 14:57:19 UTC; 33min ago
   Main PID: 782 (mjpg_streamer)
      Tasks: 4 (limit: 9079)
     Memory: 2.2M
     CGroup: /system.slice/cam_LulzMini.service
             └─782 /home/adam/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_LulzMini -r 1920x1080 -f 60 -o output_http.so -p 8002

Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: Format............: JPEG
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: TV-Norm...........: DEFAULT
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: FPS coerced ......: from 60 to 30
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: www-folder-path......: disabled
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: HTTP TCP port........: 8002
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: HTTP Listen Address..: (null)
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: username:password....: disabled
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: commands.............: enabled
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: starting input plugin input_uvc.so
Apr 06 14:57:19 octoprint mjpg_streamer[782]: MJPG-streamer [782]: starting output plugin: output_http.so (ID: 00)

paukstelis, 2022-04-06:
Do a `cat /etc/systemd/system/cam_SV04.service` and paste the output

LehmannADL, 2022-04-06:
adam@octoprint:~$ cat /etc/systemd/system/cam_SV04.service
[Unit]
Description=the OctoPi(buntu) webcam daemon with the user specified config
After=network.online.target
Wants=network.online.target

[Service]
User=adam
Environment="LD_LIBRARY_PATH=/home/adam/mjpg-streamer"
ExecStart=/home/adam/mjpg-streamer/mjpg_streamer -i "input_uvc.so -d /dev/cam_SV04 -r 1080x1920 -f 60" -o "output_http.so -p 8001"

[Install]
WantedBy=multi-user.target
adam@octoprint:~$

paukstelis, 2022-04-06:
Resolution is incorrect. Should be 1920x1080.
sudo nano /etc/systemd/system/cam_SV04.service
then reboot.

LehmannADL, 2022-04-06:
done

LehmannADL, 2022-04-06:
still no image

paukstelis, 2022-04-06:
There is something wrong, either with your camera or the parameters you have sent to it. I would start with changing the parameters to something more simple: -r 640x480 -f 5
Make sure that works first.

LehmannADL, 2022-04-06:
So i did so then i did.

adam@octoprint:~$ ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  6 17:08 /dev/cam_LulzMini -> video0

looks like it deleted the other cam, also now neither cam works at all

LehmannADL, 2022-04-06:
changed resolution 2 1280x720 cam is back on but other is not

LehmannADL, 2022-04-06:
ok so it will take a picture on octoprint with the command
http://octoprint.local:8001?action=snapshot
 will not do video

LehmannADL, 2022-04-06:
nvm it just fixed its self! THANKS FOR ALL THE HELP ENJOY THE COFFEE!

LehmannADL, 2022-04-06:
where is 

Path to FFMPEG   for octoprint?

LehmannADL, 2022-04-06:
Any chance FFMPEG running on both cameras might cause this?