paukstelis/octoprint_deploy #14: Little help fixing same serial cams.

LehmannADL, 2022-04-07:
Thanks for all your help this far, I've got the new camera and I want to basically make this as easy as possible. I assume I need to delete both cameras, run through your program add them again to each instance. Will this work?

paukstelis, 2022-04-07:
No, since you are having cameras detected by USB port, just plug it into
the same USB slot as the old cameras.

LehmannADL, 2022-04-07:
ok, ive done what youve asked and the SV04 cam will still not come online to stream

LehmannADL, 2022-04-07:
dmesg

[  506.229964] usb 5-2: USB disconnect, device number 3
[  514.144368] usb 5-2: new high-speed USB device number 4 using xhci_hcd
[  514.721669] usb 5-2: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[  514.721672] usb 5-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  514.721674] usb 5-2: Product: HDY Webcam USB
[  514.721676] usb 5-2: Manufacturer: HDY Webcam USB
[  514.721677] usb 5-2: SerialNumber: HDY Webcam USB
[  514.726479] uvcvideo: Found UVC 1.00 device HDY Webcam USB (0c45:636b)
[  514.739220] input: HDY Webcam USB: HDY Webcam USB as /devices/pci0000:00/0000:00:1c.7/0000:06:00.0/usb5/5-2/5-2:1.0/input/input19
[  514.746058] usb 5-2: 3:1: cannot get freq at ep 0x84

LehmannADL, 2022-04-07:
dmesg (LulzMini Cam)

[  654.241425] usb 2-1.6: USB disconnect, device number 4
[  654.244150] uvcvideo: Failed to resubmit video URB (-19).
[  654.248165] uvcvideo: Failed to resubmit video URB (-19).
[  654.252150] uvcvideo: Failed to resubmit video URB (-19).
[  654.256039] uvcvideo: Failed to resubmit video URB (-19).
[  654.260027] uvcvideo: Failed to resubmit video URB (-19).
[  657.281519] usb 2-1.6: new high-speed USB device number 5 using ehci-pci
[  657.479921] usb 2-1.6: New USB device found, idVendor=0c45, idProduct=636b, bcdDevice= 1.00
[  657.479924] usb 2-1.6: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  657.479926] usb 2-1.6: Product: USB 2.0 Camera
[  657.479928] usb 2-1.6: Manufacturer: Sonix Technology Co., Ltd.
[  657.479929] usb 2-1.6: SerialNumber: SN0001
[  657.482035] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:636b)
[  657.499416] input: USB 2.0 Camera: USB Camera as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/input/input20
[  657.506921] usb 2-1.6: 3:1: cannot get freq at ep 0x84

LehmannADL, 2022-04-07:
so from what im seeing its literally the same camera minus the

LulzMini Cam
[ 657.479928] usb 2-1.6: Manufacturer: Sonix Technology Co., Ltd.
[ 657.479929] usb 2-1.6: SerialNumber: SN0001

SV04 Cam
[ 514.721676] usb 5-2: Manufacturer: HDY Webcam USB
[ 514.721677] usb 5-2: SerialNumber: HDY Webcam USB

paukstelis, 2022-04-08:
I cannot tell you for certain what the issue is. It is possible that the USB bus on your computer can't handle multiple cameras. Cameras are still listed as experimental. There are way too many possibilities with cameras for me to be able to take into account all of them.

LehmannADL, 2022-04-08:
Well any chance I could get the proper code to delete the cameras? And then add them again via your software?

paukstelis, 2022-04-08:
Delete the camera lines from /etc/udev/rules.d/99-octoprint.rules
sudo systemctl stop cam_instancename.service
sudo systemctl disable cam_instancename.service
sudo rm /etc/systemd/system/cam_instancename.service
reboot
Then you can use the Add USB camera when you run octoprint deploy.

LehmannADL, 2022-04-08:
ok well now none of them work now.

**Tried systemctl status (SV04)** 

 sudo systemctl status cam_SV04

● cam_SV04.service - the OctoPi(buntu) webcam daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam_SV04.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-04-08 09:33:20 EDT; 9min ago
    Process: 786 ExecStart=/home/root/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_SV04 -r 1280X720 -f 30 -o output_http.so -p 8003 (code=exite>
   Main PID: 786 (code=exited, status=203/EXEC)

Apr 08 09:33:20 octoprint systemd[1]: Started the OctoPi(buntu) webcam daemon with the user specified config.
Apr 08 09:33:20 octoprint systemd[786]: cam_SV04.service: Failed to execute command: No such file or directory
Apr 08 09:33:20 octoprint systemd[786]: cam_SV04.service: Failed at step EXEC spawning /home/root/mjpg-streamer/mjpg_streamer: No such file or directory
Apr 08 09:33:20 octoprint systemd[1]: cam_SV04.service: Main process exited, code=exited, status=203/EXEC
Apr 08 09:33:20 octoprint systemd[1]: cam_SV04.service: Failed with result 'exit-code'.

**Tried systemctl status (LULZMINI)**

root@octoprint:~# sudo systemctl status cam_LulzMini
● cam_LulzMini.service - the OctoPi(buntu) webcam daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam_LulzMini.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-04-08 09:33:20 EDT; 13min ago
    Process: 784 ExecStart=/home/root/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_LulzMini -r 1280x720 -f 30 -o output_http.so -p 8004 (code=exited, st>
   Main PID: 784 (code=exited, status=203/EXEC)

Apr 08 09:33:20 octoprint systemd[1]: Started the OctoPi(buntu) webcam daemon with the user specified config.
Apr 08 09:33:20 octoprint systemd[784]: cam_LulzMini.service: Failed to execute command: No such file or directory
Apr 08 09:33:20 octoprint systemd[784]: cam_LulzMini.service: Failed at step EXEC spawning /home/root/mjpg-streamer/mjpg_streamer: No such file or directory
Apr 08 09:33:20 octoprint systemd[1]: cam_LulzMini.service: Main process exited, code=exited, status=203/EXEC
Apr 08 09:33:20 octoprint systemd[1]: cam_LulzMini.service: Failed with result 'exit-code'.

root@octoprint:~# cat /etc/udev/rules.d/99-octoprint.rules
KERNELS=="5-1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_SV04"
SUBSYSTEM=="tty", ATTRS{serial}=="7403530343035131C061", SYMLINK+="octo_LulzMini"

SUBSYSTEM=="video4linux", ATTRS{serial}=="HDY Webcam USB", ATTR{index}=="0", SYMLINK+="cam_SV04"
SUBSYSTEM=="video4linux", ATTRS{serial}=="SN0001", ATTR{index}=="0", SYMLINK+="cam_LulzMini"

root@octoprint:~# ls -la /dev/cam*
lrwxrwxrwx 1 root root 6 Apr  8 09:41 /dev/cam_LulzMini -> video0
lrwxrwxrwx 1 root root 6 Apr  8 09:41 /dev/cam_SV04 -> video2

paukstelis, 2022-04-08:
It look like you've run octoprint_deploy as root instead of a user based on the path. You can edit the cam service files to point to the correct path for mjpg_streamer (/home/username/mjpg-streamer/mjpg_streamer)

LehmannADL, 2022-04-08:
OK so i did a complete reinstall of server and your program. assigned each camera to each instance..... Still only one camera is streaming.

paukstelis, 2022-04-08:
Sorry. I've tried to help, but it sounds like there is any issue with your system not being able to support two cameras. You can give this a read and maybe get some insights: https://www.ideasonboard.org/uvc/faq/

paukstelis, 2022-04-08:
Now that your cameras are being detected by serial number, you can try putting them in different USB port positions. If you have more than 1 USB bus on the computer, it may fix the issue.