paukstelis/octoprint_deploy #15: Independent printers

fabiomaticus, 2022-04-18:
I have two printers with two octoprint instanze. I noticed that if one is printing and I turn off the other, automatically the active one freezes and you lose your job.

paukstelis, 2022-04-18:
This shouldn't happen and I have never seen it happen. Please provide octoprint.log files right for both instances after this happens. Are you using AUTO for detecting the serial port, or are you using the octo_[instance] name for connecting to the printers?

fabiomaticus, 2022-04-19:
Hello. I didn't select AUTO and each printer connects to its own USB instance (/dev/octo_INSTANCE)

paukstelis, 2022-04-20:
You will need to provide some logs for each of your instances to look into why this happens.

closed 2022-08-31 by paukstelis