paukstelis/octoprint_deploy #18: Camera not workin

enishi1357, 2022-04-23:
I setup correctly with the serial number recognized for both printer and camera. But when I got to http://localhost:8001/webcam/?action=stream? I get this

501: Not Implemented!
no www-folder configured

paukstelis, 2022-04-23:
There will no longer be anything at the `webcam` address via haproxy when you use octoprint_deploy. You need to connect directly to the camera instance with http://hostname.local:8001/?action=stream

enishi1357, 2022-04-23:
Tried that and the feed is just loading forever.

paukstelis, 2022-04-23:
Make sure mjpg-streamer is running: `sudo systemctl status cam_instancename`

enishi1357, 2022-04-23:
● cam_ET5X.service - the OctoPi(buntu) webcam daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam_ET5X.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-04-23 01:25:59 HST; 11h ago
   Main PID: 749 (mjpg_streamer)
      Tasks: 4 (limit: 9359)
     Memory: 984.0K
     CGroup: /system.slice/cam_ET5X.service
             └─749 /home/octouser/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_ET5X -r 640x480 -f 5 -o output_http.so -p 8002

Apr 23 01:26:00 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: TV-Norm...........: DEFAULT
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: www-folder-path......: disabled
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: HTTP TCP port........: 8002
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: HTTP Listen Address..: (null)
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: username:password....: disabled
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: commands.............: enabled
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: starting input plugin input_uvc.so
Apr 23 01:26:01 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: starting output plugin: output_http.so (ID: 00)
Apr 23 01:26:06 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: select() timeout
Apr 23 01:26:06 paul-300E4A-300E5A-300E7A-3430EA-3530EA mjpg_streamer[749]: MJPG-streamer [749]: cleaning up resources allocated by input thread

enishi1357, 2022-04-23:
is it running?

www-folder-path......: disabled

paukstelis, 2022-04-23:
Doesn't look like it based on the timeout.
what does ls -la /dev/cam_* show?

enishi1357, 2022-04-24:
lrwxrwxrwx 1 root root 6 Apr 23 01:25 /dev/cam_ET5X -> video2

enishi1357, 2022-05-17:
how do I change video2 to video0?

paukstelis, 2022-05-17:
First, paste the contents of the command: cat /etc/udev/rules.d/99-octoprint.rules

To edit the service file: sudo nano /etc/systemd/system/cam_ET5X.service
Followed by:
sudo systemctl restart cam_ET5X