paukstelis/octoprint_deploy #19: Generic instance not restarting

andraisr, 2022-05-11:
I went through your video and was able to get the initial instance up and running, then I continued on and setup another printer and a camera. You mentioned in the video that the first instance was disabled. As I saw when I opened the browser. then i proceeded to add the second printer and camera. I went back to use the first instance (generic) and it would not start. Please advise, I am not adverse to starting over if I missed a step.

Thanks in advance,
Sincerely,

Israel Andrade

paukstelis, 2022-05-11:
There is no need to use the generic instance for anything after you setup
the admin user account. It is simply a template.

andraisr, 2022-05-11:
So how do I restart the first instance. Or just make another?

paukstelis, 2022-05-11:
You just make another. The first instance is only a template, which is why it is turned off after you make the admin user.

andraisr, 2022-05-12:
Made another. it works. Thanks again. Silly question. If I do the firmware updater add on. does this get added to all instances, or just the one I loaded it on to. I also need to do the avrdude install via sudo, again, does this affect all instances or just the one I connected to with putty?

paukstelis, 2022-05-12:
Any plugin you add will be added to all instances. Updates to one instance will propagate to all of the instances. You just have to restart the instances for it to take effect.

closed 2022-06-09 by paukstelis