paukstelis/octoprint_deploy #20: Cant Access Instances

bcan1ne, 2022-05-26:
Hello,

First off, very nice and useful program!

I have installed (2) instances (with machines with SN detected) and (2) cameras. During the setup I never tried to access the instances but after finishing I tried and cannot access via IP:5001, 5002 which I noticed were the assigned ports.

I have the printers running now but plan to start over. Not sure if you have any suggestions. Thanks in advance!

paukstelis, 2022-05-26:
First thing to do is make sure the instances are running: sudo systemctl status instancename

bcan1ne, 2022-06-02:
Thanks for the help! I went ahead and started fresh and it worked.

Im not sure if it caused the issue but the first time I set it up (and was having the issues) I did add a space to the instance name.

Thanks again for the great work and help!

closed 2022-06-09 by paukstelis