paukstelis/octoprint_deploy #22: cant delete instances

LehmannADL, 2022-05-30:
trying to delete my lulzbot instance. when i select delete it just reloads on linux. can select or delete.

paukstelis, 2022-05-31:
Is it an instance that you made using the script? You can check which instances have been created with: cat /etc/octoprint_instances

thirtyxarcher, 2022-07-06:
I'm having a similar issue where I created instances but they don't show up in the Delete menu. 
[image]
[image]

paukstelis, 2022-07-06:
That's interesting. I'll have a look into it

paukstelis, 2022-07-07:
Turned out to be very simple. I used :alnum: regex when I should have used :graph: to take in account punctuation characters. Changes pushed. To update simply: `git -C octoprint_deploy pull` then try and delete instance.

closed 2022-07-07 by paukstelis