paukstelis/octoprint_deploy #23: How to Update from previous version ( before addition of ustreamer?)

SadChungus, 2022-06-07:
Tried remove command didn't work, tried pip uninstall didn't work....I'm not very good at this.

paukstelis, 2022-06-08:
`cd octoprint_deploy; git pull` will update octoprint_deploy. Keep in mind that it is not 100% compatible between updates. It would be best to do the remove command and then Prepare system again with the new version.

SadChungus, 2022-06-08:
Yes, I would like to do that, but the remove command is not compatible in the version I have, I was able to use comments to reprepare the the system, but there were indeed compatibility issues, everything is stable and running for me, 2 machines with new update, one machine with a half update.

SadChungus, 2022-06-08:
I tried the get pull, and it added the script needed for a successful remove command, thank you.

closed 2022-06-08 by SadChungus