paukstelis/octoprint_deploy #26: Idea - possibility of creating instances with no printer, but only uStreamer webcams?

kasperhvid, 2022-07-14:
So i run a printfarm where i use your script, and i love it!! However i would like to run my webcams on a seperate print, and the process of creating multiple instances of uStreamer seems complicated.

It occured to me that your script can do exactly this with proper serial # detection of the cams(very nice)

However your script will not allow me to create a "blank instance" (where i dont connect any printer but only a webcam). Is it possible to make this an option? Or is there an easy way for me to modify your script to do this?

Again, thanks so much for this genius developement!!

paukstelis, 2022-07-14:
I will consider it, but I generally hate webcams and the support they require, so it might take a lot of motivation for me to get around to it.

Building and installing ustreamer is not hard, and the service files are quite straightforward. That would be the ideal approach. There is a work around if you want run webcams on a separate machine from the printers but still use octoprint_deploy for creating ustreamer service files:

Setup the other machine the same way you would with octoprint_deploy.
Add an instance and use a free printer board or one of your other printers to do the autodetection to make a "fake" instance. It would be best to use one that does not have a serial number in this case, but it doesn't really matter.
Setup the camera for that "fake" instance as you would normally.
Then, you can simply turn off the fake instance so it won't  start on boot: `sudo systemctl disable fakeinstancename`
You can now do this with all your other cameras, using the same printer board or printer to make fake instances, but with installed cameras. You would then need to edit your other OctoPrint instances to point to the host and port of the machine you have the cameras install on.

kasperhvid, 2022-07-14:
Wow, thanks for the quick reply.

I agree that webcams are a hassle! When you mean printer board, you mean basically take a mainboard out of a printer and plug it in? (I have an old ender mainboard at home, im pretty sure those have no serial number)

Would the mainboard require 24v power in order to be detected? Or can i simply utilize the USB connection?

paukstelis, 2022-07-14:
Yes, just a free board. If it would need to be powered likely depends on the board. I suspect since most creality board can be backpowered, you would not need to supply 24V. Most of my old RAMPS-clones I use for testing do not need to be powered either.

kasperhvid, 2022-07-14:
Thanks a lot for your help. I'll mark this as closed and attempt this as soon as possible!

kasperhvid, 2022-07-15:
[image]

I tried this idea and it worked great using a dummy mainboard. Thanks!

However, my logitech C270 webcams apparently have the same serial #, so the symlinks/udev rules cause some issues

I tried to edit the udev rules in order to mimick the mainboard rule, so that the cameras are linked to a physical port - i found the part in "KERNELS==" by running udevadm monitor and unplugging/pluggin each camera"

However it doesnt work still after a reboot - uStreamer shows error "Can't access device: No such file or directory"

Any ideas about how i can fix this?

closed 2022-07-27 by kasperhvid