paukstelis/octoprint_deploy #28: Problem with one instance octoprint

fabiomaticus, 2022-07-19:
[image]

paukstelis, 2022-07-19:
You can begin by ssh'ing into the Raspberry Pi and seeing what is happening with the instance: `sudo systemctl status instancename`

fabiomaticus, 2022-07-20:
This is response:

● ninetta.service - The snappy web interface for your 3D printer
   Loaded: loaded (/etc/systemd/system/ninetta.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2022-07-19 22:17:07 BST; 3h 4min ago
  Process: 407 ExecStart=/home/pi/oprint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT} (code=exited, status=0/SUCCESS)
 Main PID: 407 (code=exited, status=0/SUCCESS)

Jul 19 22:17:07 octopi octoprint[407]: Configured folders didn't pass sanity check: Folder at /home/pi//.ninetta/uploads doesn't appear to be writable, please fix its permissions
Jul 19 22:17:07 octopi octoprint[407]: There was a fatal error starting up OctoPrint.
Jul 19 22:17:07 octopi octoprint[407]: 2022-07-19 22:17:07,181 - octoprint.settings - ERROR - Could not write test file to /home/pi//.ninetta/uploads/.testballoon.txt
Jul 19 22:17:07 octopi octoprint[407]: Traceback (most recent call last):
Jul 19 22:17:07 octopi octoprint[407]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 2385, in _validate_folder
Jul 19 22:17:07 octopi octoprint[407]:     os.remove(testfile)
Jul 19 22:17:07 octopi octoprint[407]: FileNotFoundError: [Errno 2] No such file or directory: '/home/pi//.ninetta/uploads/.testballoon.txt'
Jul 19 22:17:07 octopi octoprint[407]: 2022-07-19 22:17:07,181 - octoprint.startup - CRITICAL - Configured folders didn't pass sanity check: Folder at /home/pi//.ninetta/uploads doesn't appear to be writable, please fix its permissions
Jul 19 22:17:07 octopi octoprint[407]: 2022-07-19 22:17:07,181 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
Jul 19 22:17:07 octopi systemd[1]: ninetta.service: Succeeded.

● ninetta.service - The snappy web interface for your 3D printer
   Loaded: loaded (/etc/systemd/system/ninetta.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2022-07-19 22:17:07 BST; 3h 2min ago
  Process: 407 ExecStart=/home/pi/oprint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT} (code=exited, status=0/SUCCESS)
 Main PID: 407 (code=exited, status=0/SUCCESS)

Jul 19 22:17:07 octopi octoprint[407]: Configured folders didn't pass sanity check: Folder at /home/pi//.ninetta/uploads doesn't appear to be writable, please fix its permissions
Jul 19 22:17:07 octopi octoprint[407]: There was a fatal error starting up OctoPrint.
Jul 19 22:17:07 octopi octoprint[407]: 2022-07-19 22:17:07,181 - octoprint.settings - ERROR - Could not write test file to /home/pi//.ninetta/uploads/.testballoon.txt
Jul 19 22:17:07 octopi octoprint[407]: Traceback (most recent call last):
Jul 19 22:17:07 octopi octoprint[407]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 2385, in _validate_folder
Jul 19 22:17:07 octopi octoprint[407]:     os.remove(testfile)
Jul 19 22:17:07 octopi octoprint[407]: FileNotFoundError: [Errno 2] No such file or directory: '/home/pi//.ninetta/uploads/.testballoon.txt'
Jul 19 22:17:07 octopi octoprint[407]: 2022-07-19 22:17:07,181 - octoprint.startup - CRITICAL - Configured folders didn't pass sanity check: Folder at /home/pi//.ninetta/uploads doesn't appear to be writable, please fix its permissions
Jul 19 22:17:07 octopi octoprint[407]: 2022-07-19 22:17:07,181 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
Jul 19 22:17:07 octopi systemd[1]: ninetta.service: Succeeded.

paukstelis, 2022-07-20:
provide output of: `cat /etc/systemd/system/ninetta.service`

and: `ls -la ~/.ninetta/`

fabiomaticus, 2022-07-20:
Now the instance is running, without having touched anything: this is the third time it has happened.

fabiomaticus, 2022-07-20:
pi@octopi:~ $ cat /etc/systemd/system/ninetta.service
[Unit]
Description=The snappy web interface for your 3D printer
After=network.online.target
Wants=network.online.target

[Service]
Environment="PORT=5001"
Environment="BASEDIR=/home/pi//.ninetta"
Environment="CONFIGFILE=/home/pi//.ninetta/config.yaml"
User=pi
ExecStart=/home/pi/oprint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT}

[Install]
WantedBy=multi-user.target

fabiomaticus, 2022-07-20:
total 76
drwxr-xr-x 14 pi pi 4096 Jul 20 01:29 .
drwxr-xr-x 14 pi pi 4096 Apr 15 12:04 ..
-rw-r--r--  1 pi pi 4327 Jul 20 01:26 config.backup
-rw-r--r--  1 pi pi 4340 Jul 20 01:29 config.yaml
drwxr-xr-x 10 pi pi 4096 Jul 20 06:18 data
drwxr-xr-x  3 pi pi 4096 Jul 20 01:26 generated
drwxr-xr-x  2 pi pi 4096 Jul 20 01:29 logs
drwxr-xr-x  2 pi pi 4096 Jul 20 01:26 plugins
drwxr-xr-x  2 pi pi 4096 Jul 20 01:26 printerProfiles
drwxr-xr-x  3 pi pi 4096 Jul 20 01:26 scripts
drwxr-xr-x  2 pi pi 4096 Jul 20 01:26 slicingProfiles
drwxr-xr-x  3 pi pi 4096 Jul 20 01:26 timelapse
drwxr-xr-x  3 pi pi 4096 Jul 20 01:26 translations
drwxr-xr-x 30 pi pi 4096 Jul 20 01:26 uploads
-rw-------  1 pi pi  314 Apr 15 10:30 users.yaml
drwxr-xr-x  2 pi pi 4096 Jul 20 01:26 virtualSd
drwxr-xr-x  2 pi pi 4096 Jul 20 01:26 watched

paukstelis, 2022-07-20:
Based on the error message and the behavior, I would think your SD card is going bad.

fabiomaticus, 2022-07-20:
Please how can I properly clone the SD card with Windows?

fabiomaticus, 2022-07-21:
I have change SD: same iusse.
One instance works and the other does not work.

paukstelis, 2022-07-21:
How about trying: `sudo systemctl restart ninetta; sudo systemctl status ninetta`

The error message clearly indicates there is an issue with trying to write a test file when OctoPrint starts. That has nothing to do with octoprint_deploy script, so I may not be able to figure out the issue. You could try posting the error message on the OctoPrint discord or the community forum.

fabiomaticus, 2022-07-21:
pi@octopi:~ $ sudo systemctl status ninetta
● ninetta.service - The snappy web interface for your 3D printer
   Loaded: loaded (/etc/systemd/system/ninetta.service; enabled; vendor preset:
   Active: active (running) since Thu 2022-07-21 06:12:54 BST; 44s ago
 Main PID: 1335 (octoprint)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/ninetta.service
           └─1335 /home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serv

Jul 21 06:13:31 octopi octoprint[1335]: 2022-07-21 06:13:31,441 - octoprint.serv
Jul 21 06:13:31 octopi octoprint[1335]: 2022-07-21 06:13:31,441 - octoprint.acce
Jul 21 06:13:31 octopi octoprint[1335]: 2022-07-21 06:13:31,509 - octoprint.serv
Jul 21 06:13:32 octopi octoprint[1335]: 2022-07-21 06:13:32,183 - octoprint.plug
Jul 21 06:13:32 octopi octoprint[1335]: 2022-07-21 06:13:32,220 - octoprint.serv
Jul 21 06:13:32 octopi octoprint[1335]: 2022-07-21 06:13:32,257 - octoprint.serv
Jul 21 06:13:32 octopi octoprint[1335]: 2022-07-21 06:13:32,257 - octoprint.acce
Jul 21 06:13:33 octopi octoprint[1335]: 2022-07-21 06:13:33,573 - octoprint.serv
Jul 21 06:13:34 octopi octoprint[1335]: 2022-07-21 06:13:34,601 - octoprint.plug
Jul 21 06:13:34 octopi octoprint[1335]: 2022-07-21 06:13:34,722 - octoprint.plug
lines 1-18/18 (END)

fabiomaticus, 2022-07-21:
"sudo systemctl restart ninetta"

The instance has started working again