paukstelis/octoprint_deploy #29: Problem wiht Add New instance

fabiomaticus, 2022-07-25:
I want to add a new instance. I run the command sudo octoprint_deploy / octoprint_deploy.sh
I click on "Add Instance", but it repeats the question over and over again.
[image]

paukstelis, 2022-07-25:
This is because your octoprint_deploy version is newer than the version used to do the system prepare. The solution is to make a file called octoprint_deploy in /etc that contains the necessary information.

The contents should look like this:
```
type: linux/octopi
streamer: ustreamer/mjpg-streamer
haproxy: true/false
```
where you select the correct option for each line

fabiomaticus, 2022-07-25:
can you help me? I don't know what to do!

paukstelis, 2022-07-25:
Yes, I can try. First, are you using linux (ubuntu, etc.) or octopi?

fabiomaticus, 2022-07-25:
Octopi

paukstelis, 2022-07-25:
OK, then do the following from an ssh/putty session on the RPi:
`sudo nano /etc/octoprint_deploy`
Then paste the following:
```
type: octopi
streamer: mjpg-streamer
haproxy: true
```
then Ctrl+X to save the file.

fabiomaticus, 2022-07-25:
Now, sudo octoprint_deploy / octoprint_deploy.sh

paukstelis, 2022-07-25:
Yes.

fabiomaticus, 2022-07-25:
^I have this problem
[image]

paukstelis, 2022-07-25:
That is also an issue from preparing the system with an older version.

`sudo apt install  uuid-runtime` in the terminal should fix.

You could remove the instance and reinstall it if you want  to, but it will likely work without issue (unless you use home assistant or or something that requires unique ids).

fabiomaticus, 2022-07-25:
Many thanks. It works perfectly without sending new commands. I offer you a coffee. Many thanks again.

closed 2022-07-25 by paukstelis