paukstelis/octoprint_deploy #31: OctoPrint Not Working...

XeroZar, 2022-08-04:
I install everything with 1 minor hitch, see attached text file, on Ubuntu 22.04 LTS. I rebooted and try to access OctoPrint but no luck, I checked for any OctoPrint services and nothings there. Tried to start one but  it says "Failed to start octoprint.service: Unit octoprint.service not found." Any help would be appreciated
[Error.txt](https://github.com/paukstelis/octoprint_deploy/files/9263482/Error.txt)
.

paukstelis, 2022-08-04:
Did you make a new instance after creating the template?
There is no octoprint.service when you use octoprint_deploy, instead, you make an instance using the script and then those service files are named after the instance (e.g. printer1.service).

XeroZar, 2022-08-04:
Yes, I named mine Ender3Pro...

paukstelis, 2022-08-04:
The you can check it is working from a terminal/ssh session with: `sudo systemctl status Ender3Pro`
You can also verify the information with: `cat /etc/systemd/system/Ender3Pro.service`

XeroZar, 2022-08-04:
I'll try that thank you... Also I was wondering there was a recent update for python3-distupgrade and ubunt-releaseupgrader-core, would these effect the install process?

paukstelis, 2022-08-04:
No, those should not impact the installation.
If you want more direct support, join the Discord channel in the main page/README.

XeroZar, 2022-08-04:
Ok cool, have the link by any chance?

paukstelis, 2022-08-04:
https://discord.gg/6vgSjgvR6u

closed 2022-08-31 by paukstelis