paukstelis/octoprint_deploy #34: Prusa Mini not being detected but dmesg shows the printer

ToledoRayburn, 2022-08-28:
I tried unplugging and replugging .    The generic instance can talk to it on ttyAMC0

Dmesg output
Device= 0.02
[   26.091403] usb 1-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  131.686487] usb 1-6: new full-speed USB device number 7 using xhci_hcd
[  131.840654] usb 1-6: New USB device found, idVendor=2c99, idProduct=000c, bcdDevice= 2.00
[  131.840665] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  131.840671] usb 1-6: Product: Original Prusa MINI
[  131.840675] usb 1-6: Manufacturer: Prusa Research (prusa3d.com)
[  131.840679] usb 1-6: SerialNumber: CZPX2622X017XC85868
[  131.995714] usbcore: registered new interface driver cdc_acm
[  531.201970] usb 1-6: USB disconnect, device number 7
[  536.110562] usb 1-6: new full-speed USB device number 8 using xhci_hcd
[  536.260440] usb 1-6: New USB device found, idVendor=2c99, idProduct=000c, bcdDevice= 2.00
[  536.260453] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  536.260459] usb 1-6: Product: Original Prusa MINI
[  536.260463] usb 1-6: Manufacturer: Prusa Research (prusa3d.com)
[  536.260468] usb 1-6: SerialNumber: CZPX2622X017XC85868

paukstelis, 2022-08-28:
So in octoprint_deploy just times out? I've tested with a Mini with no issues, and based on that dmesg output it should detect the serial number.

ToledoRayburn, 2022-10-11:
It just times out... if I do the usb check, it also times out.   I am doing an Unbuntu install if that helps.

paukstelis, 2022-10-11:
You can try updating octoprint_deploy (git -C octoprint_deploy pull). It
now uses dmesg for detection, so if it actually is detected, it will work.

davidzwa, 2023-07-31:
Can confirm here that Prusa Mini is detected on latest master. You could consider closing this issue.