paukstelis/octoprint_deploy #35: 1 of 3 instances cannot connect to camera

Maksiator, 2022-08-31:
Hello,
I configured 3 instances of octoprint (prusa i3 mk3s+) and 3 logitech cameras. 2 cameras works fine with instance number 1 and 3, but camera 2 can't connect to instance number 2. I tried restarting service using sudo systemctl restart cam_prusa-5215 and then sudo systemctl status cam_prusa-5215 for check if it's working. 

```
● cam_prusa-5215.service - the OctoPi(buntu) webcam daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam_prusa-5215.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-08-31 15:03:30 UTC; 11s ago
   Main PID: 1656 (mjpg_streamer)
      Tasks: 3 (limit: 18968)
     Memory: 1.4M
     CGroup: /system.slice/cam_prusa-5215.service
             └─1656 /home/max/mjpg-streamer/mjpg_streamer -i input_uvc.so -d /dev/cam_prusa-5215 -r 1920x1080 -f 30 -o output_http.so -p 8002

Aug 31 15:03:30 kanciapa mjpg_streamer[1656]:  o: HTTP Listen Address..: (null)
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]:  o: username:password....: disabled
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]:  o: commands.............: enabled
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: www-folder-path......: disabled
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: HTTP TCP port........: 8002
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: HTTP Listen Address..: (null)
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: username:password....: disabled
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: commands.............: enabled
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: starting input plugin input_uvc.so
Aug 31 15:03:30 kanciapa mjpg_streamer[1656]: MJPG-streamer [1656]: starting output plugin: output_http.so (ID: 00)
```
And as you can see it's working, but octoprint instance can't load view.

paukstelis, 2022-08-31:
My first thought is that you are probably running out of USB bandwidth on that bus. If your cameras have serial numbers, try putting them in different USB ports to spread the load. Ideally, you'll have multiple USB busses on the system and you would want to have only one camera on each.

Maksiator, 2022-08-31:
That was it. I ran out of USB bus. Thank you!

closed 2022-08-31 by paukstelis