paukstelis/octoprint_deploy #37: Camera stream lagging

Prototype7813, 2022-09-25:
Hi ! 

I did a setup with Ustream (and like the script was telling after the install , I put 0 for the FPS) and the stream seems to run at 2-5 fps.

Is it normal ? Do I need to change for mjpeg-streamer? 

The camera is a logitech c920 and it's plug into a USB hub. 

Thank you for your help

paukstelis, 2022-09-25:
first I would check the service file:
`cat /etc/systemd/system/cam_INSTANCENAME.service`
and then see if there is any info about how it is running:
`sudo systemctl status cam_INSTANCENAME -l --no-pager`

Prototype7813, 2022-09-25:
Seems like my client disconnected and reconnect. I'm also having an issue with the ssh. The command line is awful to use. 

I'm putting the log at the end. With htop, I see that the load average is at 0.5-0.6. Ram is not even at 500 MB and I'm also using an SSD.

I didn't have a camera before but I didn't have this issue with accessing octoprint before. 

Sep 25 18:31:27 octoprintalpha ustreamer[401]: -- INFO  [87757.812      http] -- HTTP: Disconnected client: [10.0.0.9]:26085, id=952a6c6934f65593, Resource temporarily unavailable (reading,eof); clients now: 0
Sep 25 18:35:44 octoprintalpha ustreamer[401]: -- INFO  [88014.421      http] -- HTTP: Registered client: [10.0.0.9]:1087, id=56ea63de6490fef; clients now: 1
Sep 25 18:38:13 octoprintalpha ustreamer[401]: -- INFO  [88163.256      http] -- HTTP: Disconnected client: [10.0.0.9]:1087, id=56ea63de6490fef, Resource temporarily unavailable (reading,eof); clients now: 0
Sep 25 18:38:22 octoprintalpha ustreamer[401]: -- INFO  [88172.881      http] -- HTTP: Registered client: [10.0.0.9]:1047, id=1b52430160db796d; clients now: 1
Sep 25 18:55:07 octoprintalpha ustreamer[401]: -- INFO  [89177.317      http] -- HTTP: Disconnected client: [10.0.0.9]:1047, id=1b52430160db796d, Resource temporarily unavailable (reading,eof); clients now: 0
Sep 25 18:59:42 octoprintalpha ustreamer[401]: -- INFO  [89452.770      http] -- HTTP: Registered client: [10.0.0.9]:9946, id=74171fb6418268fc; clients now: 1
Sep 25 19:00:15 octoprintalpha ustreamer[401]: -- INFO  [89485.602      http] -- HTTP: Disconnected client: [10.0.0.9]:9946, id=74171fb6418268fc, Resource temporarily unavailable (reading,eof); clients now: 0
Sep 25 19:18:31 octoprintalpha ustreamer[401]: -- INFO  [90581.709      http] -- HTTP: Registered client: [10.0.0.9]:12868, id=2a4662f708fddf59; clients now: 1
Sep 25 19:18:38 octoprintalpha ustreamer[401]: -- INFO  [90589.147      http] -- HTTP: Registered client: [10.0.0.9]:1056, id=3f89fb800a973ac; clients now: 2
Sep 25 19:28:29 octoprintalpha ustreamer[401]: -- INFO  [91179.953      http] -- HTTP: Disconnected client: [10.0.0.9]:1056, id=3f89fb800a973ac, Resource temporarily unavailable (reading,eof); clients now: 1

paukstelis, 2022-09-26:
It is not clear to me where that log is coming from.
In order to get a better idea of what is going on you'd need to sort out the ssh issues and run the commands I previously sent.

Prototype7813, 2022-09-26:
it's from the ssh.

paukstelis, 2022-09-26:
what command generated that log?

Prototype7813, 2022-09-26:
Sorry , i edit my last comment at the last minute . Second command ! 

Thanks for your help