paukstelis/octoprint_deploy #40: Camera with ribbon cable how it works?

djepsylon, 2022-11-05:


paukstelis, 2022-11-05:
If you are using octoprint_deploy it must be setup manually.

djepsylon, 2022-11-05:
okay lol but how? sorry to ask this but I've never had to do it so I have no idea of the method and all the research I've done tells me everything and its opposite. Is the address of the stream to be entered? what is it?

djepsylon, 2022-11-05:
when i put "/webcam/?action=stream" it dont work and when i m typing in ssh "vcgencmd get_camera" i have been "supported=1 detected=1"
can you help me to do that?

paukstelis, 2022-11-05:
It is more complicated than that and to help I need to know more about how you have things installed. Can you join the Discord? It is easier to provide support there.

djepsylon, 2022-11-05:
yeah i have discord im on the octoprint server my pseudo is spockncc1701a can you mp me?