paukstelis/octoprint_deploy #42: Multiple Webcams W/ Hub

BoostMonster, 2022-11-18:
I have a powered hub plugged into a mini pc running ubuntu and when I try to add other webcams it gives me a warning that a camera with an identical serial number was detected and that it will likely cause issues, using mjpg-streamer how do I remedy this? when one camera is on the other stream just says loading, I'm trying to run 4 USB webcams through a hub for 4 separate octoprint instances, my printers are also on another hub and they don't seem to have any connection issues

closed 2022-11-18 by BoostMonster

FireMarshmellow, 2024-01-28:
Did you get this to work? have the same issue.

paukstelis, 2024-01-28:
Duplicate serial numbers for cameras can be fixed by removing the existing cameras from octoprint_deploy, then re-running octoprint_deploy with the noserial option: `sudo octoprint_deploy/octoprint_deploy.sh noserial`

FireMarshmellow, 2024-01-29:
That just exits when i try to add a camera
```
No camera was detected during the detection period.
Try again or try a different camera.
OctoprintPI@OctoprintPI:~$
```

paukstelis, 2024-01-29:
That means your camera is not recognized. It is possible that it is not a v4l camera.
You can try:
`sudo dmesg -C`
plug camera in with USB
`sudo dmesg`
Provide the output of that last command.

FireMarshmellow, 2024-01-29:
Probably unnecessary but I provided the output from both the cameras, Thank you for helping me out with this!
camera 1
```
[155879.724492] usb 1-1.1: new high-speed USB device number 43 using xhci_hcd
[155880.118808] usb 1-1.1: New USB device found, idVendor=1bcf, idProduct=2701, bcdDevice= 0.05
[155880.118855] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[155880.118875] usb 1-1.1: Product: HD 720P webcam
[155880.118892] usb 1-1.1: Manufacturer: Sunplus IT Co
[155880.118907] usb 1-1.1: SerialNumber: AN20200706001
[155880.344269] usb 1-1.1: Found UVC 1.00 device HD 720P webcam (1bcf:2701)
[155880.459368] usb 1-1.1: Warning! Unlikely big volume range (=4096), cval->res is probably wrong.
[155880.459406] usb 1-1.1: [5] FU [Mic Capture Volume] ch = 1, val = 0/4096/1
```
camera 2
```
[156001.233288] usb 1-1.2: new high-speed USB device number 44 using xhci_hcd
[156001.627574] usb 1-1.2: New USB device found, idVendor=1bcf, idProduct=2701, bcdDevice= 0.05
[156001.627620] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[156001.627639] usb 1-1.2: Product: HD 720P webcam
[156001.627656] usb 1-1.2: Manufacturer: Sunplus IT Co
[156001.627672] usb 1-1.2: SerialNumber: AN20200706001
[156001.852983] usb 1-1.2: Found UVC 1.00 device HD 720P webcam (1bcf:2701)
[156001.967953] usb 1-1.2: Warning! Unlikely big volume range (=4096), cval->res is probably wrong.
[156001.967985] usb 1-1.2: [5] FU [Mic Capture Volume] ch = 1, val = 0/4096/1
```

paukstelis, 2024-01-29:
What operating system are you using?
There seems to be a normal kernel line that is missing in your output. You
should have something along the lines of:
input: UVC Camera (046d:0825) as
/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/input/input39

FireMarshmellow, 2024-01-29:
Ubuntu 23.10 (GNU/Linux 6.5.0-1009-raspi aarch64)

FireMarshmellow, 2024-01-30:
Tried again with Debian, tried with noserial but still nope. 
```
[ 1552.782002] usb 1-1.1: new high-speed USB device number 12 using xhci_hcd
[ 1553.176528] usb 1-1.1: New USB device found, idVendor=1bcf, idProduct=2701, bcdDevice= 0.05
[ 1553.176563] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1553.176577] usb 1-1.1: Product: HD 720P webcam
[ 1553.176589] usb 1-1.1: Manufacturer: Sunplus IT Co
[ 1553.176601] usb 1-1.1: SerialNumber: AN20200706001
[ 1553.401824] usb 1-1.1: Found UVC 1.00 device HD 720P webcam (1bcf:2701)
[ 1553.421798] usb 1-1.1: Failed to query (GET_INFO) UVC control 6 on unit 1: -32 (exp. 1).
[ 1553.424366] usb 1-1.1: Failed to query (GET_INFO) UVC control 8 on unit 1: -32 (exp. 1).
[ 1553.448190] input: HD 720P webcam: HD 720P webcam as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/input/input6
[ 1553.514544] usb 1-1.1: Warning! Unlikely big volume range (=4096), cval->res is probably wrong.
[ 1553.514561] usb 1-1.1: [5] FU [Mic Capture Volume] ch = 1, val = 0/4096/1
[ 1553.640191] usb 1-1.1: Failed to query (GET_CUR) UVC control 8 on unit 1: -32 (exp. 1).
[ 1553.642780] usb 1-1.1: Failed to query (GET_CUR) UVC control 8 on unit 1: -32 (exp. 1).
[ 1553.645468] usb 1-1.1: Failed to query (GET_CUR) UVC control 8 on unit 1: -32 (exp. 1).
[ 1553.648168] usb 1-1.1: Failed to query (GET_CUR) UVC control 8 on unit 1: -32 (exp. 1).
OctoprintPI@OctoprintPI:~ $
```

paukstelis, 2024-01-30:
It should work.  You do have the correct line: `...as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/input/input6`
Make sure you are plugging in the camera after it tells you do so. It firsts ask you to verify that is unplugged, then you hit Y, then you plug in the camera.

FireMarshmellow, 2024-02-01:
I did nothing and it worked, The worst kind of success. Thanks for all the help!

paukstelis, 2024-02-01:
> I did nothing and it worked, The worst kind of success. Thanks for all the help!
> 
> Edit: I restarted the pi, one of the cameras stopped working The other camera was swapped from the other printer.

Sounds like you did not use the 'noserial' option.

FireMarshmellow, 2024-02-01:
Okay everything works fine now, When I tried it before it just quit When I tried adding a camera. But everything seems stable now, Accept the camera feeds don't show up in home assistant, But that's a minor problem. Once again thank you for all the help ❤️