paukstelis/octoprint_deploy #49: Camera tied to USB Port

alonks1234, 2023-01-03:
I don't know if I am alone in this problem, but I have several webcams with the same udev serial ID and am getting the warning that this will cause instability (and looking at the udev rules, I can see why). It seems like allowing a camera to be tied to a specific hardware port would solve this, which already exists for the printers. Any reason this is not implemented?

paukstelis, 2023-01-04:
It is implemented, it is not the default behavior, however. It is much more convenient if your cameras do have different serial numbers to be able to plug them in wherever. In your case, you want to run the script with the noserial argument: `sudo octoprint_deploy/octoprint_deploy.sh noserial`

This will skip detection of serial number for cameras and force it to use the USB port address.

alonks1234, 2023-01-04:
Got it, thanks!

closed 2023-01-04 by alonks1234