paukstelis/octoprint_deploy #52: How to remove a camera instance?

aslabsalbeh, 2023-01-19:
I might have added 2 cameras by mistake. When I installed "Camera Settings" plugin I get two cameras to select from "cedrus" and the Logitech C920 (which is the one I'm using). However the plugin keeps defaulting back to cedrus, and the focus/zoom settings get ruined. 

Looking at the running services I noticed that I have cam and cam2. Cam one status says "Active: failed" while cam2 is activing and running.

How do I delete cam service? Thanks!
[image]

paukstelis, 2023-01-19:
It looks like you have effectively stopped and disabled it with the commands you show. If you no longer want the cam_Ender3V2Neo.service, simply delete the file and reboot (or do a systemctl daemon-reload).

aslabsalbeh, 2023-01-19:
Thank you. Do you mind guiding me on how to delete the file & where is located.

I'm not an expert in linux & I was trying different commands I found here & there :)

Cheers

paukstelis, 2023-01-19:
sudo rm /etc/systemd/system/cam_Ender3V2Neo.service; sudo reboot