paukstelis/octoprint_deploy #53: PrettyGcode plugin not working

swheettaos, 2023-02-03:
So I got everything working after created a new instance using your script.  Some plugins had to be reinstalled (I had a few).. that being said, the only thing I can't get working is the prettygcode plugin.  I really like this plugin so I hope I can figure out whats going on.  When I print it just doen't show the gcode, and has a NaN on the scroll bar.  I tried removing and reinstalling it but no difference.  The plugin is this one and I'm using the latest release https://github.com/Kragrathea/OctoPrint-PrettyGCode anything for me to try?

paukstelis, 2023-02-03:
It works perfectly fine for me with my octoprint_deploy setup. Have you rebooted after installing it?

stephenwheet, 2023-02-03:
Multiple times. I did link the instances for gcode uploads. I wonder if that is a problem.  I changed my mind and I thought I deleted the shared upload folder.  So I might have broke it or it’s messing with the plug-in?  I do I make sure the linked uploads is disabled.

paukstelis, 2023-02-03:
Tough to say. You need to go through each instance and in the settings change the OctoPrint->Folders paths.

paukstelis, 2023-02-03:
> Ok I’ll try that. You don’t have the shared gcode folder right? Stephen
> […](#)

No, not using a shared folder in the case where I tested.

swheettaos, 2023-02-03:
nope that wasn't it .. bummer .. still not working .. verified the location for uploads was under the instance folder location.  I just can't get it working.  Completely uninstalled .. rebooted .. reinstalled, rebooted .. same thing.. doesn't show like it used to.

paukstelis, 2023-02-03:
You might check if you have errors in the browser console. Use Developer Tools, typically ctrl+shift+I.

paukstelis, 2023-02-03:
OK. I believe I have figured out the problem (though no solution as of yet). The issue is with haproxy. I think prettygcode is using some hardwired links that get broken by haproxy. If you connect to your instance(s) using the ip:port (rather than /instancename/) I think it will work.

paukstelis, 2023-02-03:
Ok. Known (old) issue it appears: https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/132
Perhaps the developer has abandoned the plugin.

stephenwheet, 2023-02-03:
Ok, are you not using haproxy? You said it’s working for you.

paukstelis, 2023-02-03:
I have haproxy setup, but my bookmarks are rather old and still point to ip:port, so it worked fine.

jneilliii, 2023-02-03:
You guys could try this patched version using relative paths instead of root paths. 

Copy/Paste the following URL in plugin manager > get more > ...from URL and click install.

```
https://github.com/jneilliii/OctoPrint-PrettyGCode/archive/refs/heads/patch-2.zip
```

swheettaos, 2023-02-03:
Woot!! worked like a champ!!  Thank you!

jneilliii, 2023-02-03:
Thanks for confirming. I submitted the change as a PR to the plugin's repository, but there were others that have been sitting in their a while, so seems plugin may be abandoned.

swheettaos, 2023-02-03:
> Thanks for confirming. I submitted the change as a PR to the plugin's repository, but there were others that have been sitting in there a while, so seems plugin may be abandoned.

Thats a bummer .. it's a great plugin.