paukstelis/octoprint_deploy #57: Armbian Support?

dodolcool, 2023-02-12:
Hello
im running octoprint in Armbian debian buster as alternate of raspberry pi. the question is can i use this deployment on armbian?

ps:
to be honest im not tried this before because i only have 1 device and it running 24/7.

paukstelis, 2023-02-12:
Yes. It works with Armbian.

dodolcool, 2023-02-12:
ok i will give it a shot. anyway i have some stupid question i hope you can answer it. my board only have 2 usb port attached in the board. if i want to connect lets say 6 printer + webcam, can i use some usb hub/usb port extention? or there is another correct way to it? i have seen your video but only mention about installation not mention about hardware.

KeithCu, 2023-12-29:
@dodolcool: Yes, you just need a USB hub.

paukstelis, 2023-12-30:
> ok i will give it a shot. anyway i have some stupid question i hope you can answer it. my board only have 2 usb port attached in the board. if i want to connect lets say 6 printer + webcam, can i [...]

6 printers you can probably do, but not webcams. Webcams use a lot of USB bandwidth.