paukstelis/octoprint_deploy #59: Reconfigure default template

guyou, 2023-02-12:
New to Octoprint & octoprint_deploy, I did some test&retry loops.

Does it make sense to add a main menu's entry in order to reconfigure the default template? I did not find one.

My idea: after first run, I wish to reconfigure the default template because I found many useful settings for my next printers. Thus, I select the "Reconfigure template" entry: it will start the default template (on 5000) and wait I confirm I made my changes and finally stop the service.

PS: I'm quite comfortable with bash so I can propose a patch for that.

paukstelis, 2023-02-12:
I'll think about this. One idea I have been toying with is doing a "Sync Settings" option. Like the "Sync User" or "Share Uploads" options, it would allow the user to pick settings from an instance that they want transferred to all instances (including template). This has advantages over just modifying the template.

guyou, 2023-02-12:
As a big noob, I'm not able to evaluate advantages / inconvenients of both solutions.
I can just testify that after playing with my first intance, I have some ideas for the template in order to have a better starting point for future instances.