paukstelis/octoprint_deploy #60: secondary camera file wasn't created

Millerboy3, 2023-02-16:
I added a second usb camera. It never created the cam2_* file for me to use. I've done this addition a few times. Any thoughts?
/dev/cam2_* doesn't exist.

```
● cam2_EnderM3D.service - the OctoPi(buntu) ustreamer daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam2_EnderM3D.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-02-16 00:52:30 EST; 28s ago
   Main PID: 8213 (main)
      Tasks: 3 (limit: 4307)
     Memory: 448.0K
        CPU: 89ms
     CGroup: /system.slice/cam2_EnderM3D.service
             └─8213 /home/fakepi/ustreamer/ustreamer -d /dev/cam2_EnderM3D -s 0.0.0.0 -m MJPEG -r 1280x720 -f 0 -p 8002 >

Feb 16 00:52:30 multipi3D systemd[1]: Started the OctoPi(buntu) ustreamer daemon with the user specified config.
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.868      main] -- Starting PiKVM uStreamer 5.37 ...
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.868      main] -- Using internal blank placeholder
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.869      main] -- Listening HTTP on [0.0.0.0]:8002
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.869    stream] -- Using V4L2 device: /dev/cam2_EnderM3D
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.869    stream] -- Using desired FPS: 0
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.869      http] -- Starting HTTP eventloop ...
Feb 16 00:52:30 multipi3D ustreamer[8213]: =============================================================================>
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- ERROR [100567.870    stream] -- Can't access device: No such file or direc>
Feb 16 00:52:30 multipi3D ustreamer[8213]: -- INFO  [100567.870    stream] -- Waiting for the device access ...

```

Millerboy3, 2023-02-16:
It's possible they are somehow detected as the same camera. They are both the same model.  so does that mean I need to run the -noserial option?

Millerboy3, 2023-02-16:
tried the -noserial option this didn't change anything. It didn't create the second camera file.

paukstelis, 2023-02-16:
What do you get with: `cat /etc/udev/rules.d/99-octoprint.rules`?

Millerboy3, 2023-02-16:
> What do you get with: `cat /etc/udev/rules.d/99-octoprint.rules`?

SUBSYSTEM=="tty", ATTRS{serial}=="0700B016AF1E90C55AABAC1AF50020C3", SYMLINK+="o                       cto_EnderM3D"
SUBSYSTEM=="video4linux", ATTRS{serial}=="DCF533EF", ATTR{index}=="0", SYMLINK+=                       "cam_EnderM3D"
SUBSYSTEM=="video4linux", ATTRS{serial}=="311D28AF", ATTR{index}=="0", SYMLINK+=                       "cam2_EnderM3D"
fakepi@multipi3D:~$

Millerboy3, 2023-02-16:
so just a few minutes ago cam2_EnderM3D showed up but I can't access it. 

EDIT: just did a status and noticed this..
`
● cam2_EnderM3D.service - the OctoPi(buntu) ustreamer daemon with the user specified config
     Loaded: loaded (/etc/systemd/system/cam2_EnderM3D.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-02-16 03:07:11 EST; 3s ago
   Main PID: 9857 (main)
      Tasks: 3 (limit: 4307)
     Memory: 616.0K
        CPU: 25ms
     CGroup: /system.slice/cam2_EnderM3D.service
             └─9857 /home/fakepi/ustreamer/ustreamer -d /dev/cam2_EnderM3D -s 0.0.0.0 -m MJPEG -r 1920x1080 -f 0 -p 8002 --de

Feb 16 03:07:11 multipi3D systemd[1]: Started the OctoPi(buntu) ustreamer daemon with the user specified config.
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.990      main] -- Starting PiKVM uStreamer 5.37 ...
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.990      main] -- Using internal blank placeholder
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.991      main] -- Listening HTTP on [0.0.0.0]:8002
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.991    stream] -- Using V4L2 device: /dev/cam2_EnderM3D
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.991    stream] -- Using desired FPS: 0
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.991      http] -- Starting HTTP eventloop ...
Feb 16 03:07:11 multipi3D ustreamer[9857]: ================================================================================
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- ERROR [108647.992    stream] -- Can't access device: Permission denied
Feb 16 03:07:11 multipi3D ustreamer[9857]: -- INFO  [108647.992    stream] -- Waiting for the device access ...`

so it has finally created the file but says access denied?

paukstelis, 2023-02-16:
I would just reboot the whole device. You can also try moving the USB port the camera is in.

Millerboy3, 2023-02-16:
I have done that as well.  I'll try again. I have rebooted multiple times, I have edited the udev rules, changed resolutions, and changed camera names/ports. all come back to this in the end.

paukstelis, 2023-02-16:
It is not a problem with the script. It is doing what it is supposed to. This is a hardware issue and this is one of the reasons cameras will always be an experimental feature.