paukstelis/octoprint_deploy #65: ustreamer not running on startup

JimmyRigIt, 2023-02-23:
Ustreamer installs correctly as i can get it running manually however i must do this everytime the computer reboots. Would be amazing if it started on a service that was installed via the script!

paukstelis, 2023-02-23:
Did you go through the 'Add Camera' step? That makes a service file that automatically starts on boot.

JimmyRigIt, 2023-02-25:
Yep, when creating the printer. Plugged in the printer first to get the usb port, then clicked yes for the camera and plugged that in. After restarting the PC The camera doesnt boot until manually booting ustreamer via ssh. 

Running the latest stable release of debian 11 on a bmax mini pc with a n3350 processor

Thanks for the response!

paukstelis, 2023-02-25:
What command are you giving to start ustreamer?

What is the output of `ls -la /etc/systemd/system/cam*`?

JimmyRigIt, 2023-03-03:
**To start ustreamer cd to folder then type,** 

user@computer:~/ustreamer$ ls
docs  janus  LICENSE  linters  Makefile  man  pkg  python  README.md  src  tools  ustreamer  ustreamer-dump
user@computer:~/ustreamer$ ./ustreamer --port 8001 --host 192.168.1.1

**The output is no such file/folder exists. Cd to /etc/systemd/system and this was the list**

user@computer:/etc/systemd/system$ ls

home.service                        default.target.wants     multi-user.target.wants      sshd.service
dbus-fi.w1.wpa_supplicant1.service          display-manager.service  network-online.target.wants  suspend.target
dbus-org.freedesktop.Avahi.service          getty.target.wants       octoprint_default.service    sysinit.target.wants
dbus-org.freedesktop.ModemManager1.service  graphical.target.wants   printer.target.wants         syslog.service
dbus-org.freedesktop.nm-dispatcher.service  hibernate.target         sleep.target                 timers.target.wants
dbus-org.freedesktop.timesync1.service      hybrid-sleep.target      sockets.target.wants         ustreamer@.service

paukstelis, 2023-03-03:
That output looks odd. Can you join the Discord server linked in the README file? It is easier to provide real-time support there.