paukstelis/octoprint_deploy #66: Restart Fails from menu

reuter, 2023-03-02:
I only have 1 instance installed, but when I try to restart that instance using the Octoprint menu, it fails, and I get the following in the logs

> 2023-03-02 19:19:23,700 - octoprint.server.api.system - INFO - Performing command for core:reboot: sudo shutdown -r now
2023-03-02 19:19:24,217 - octoprint.server.api.system - WARNING - Command for core:reboot failed with return code 1:
! 
! STDOUT:
! 
! 
! STDERR:
! sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper
! 
! sudo: a password is required

paukstelis, 2023-03-02:
That command is not to restart the instance, but to restart the host. You can change the command in octoprint to `sudo reboot` which does the same thing.

reuter, 2023-03-02:
Seems like even the restart instance doesn't work either.

> 2023-03-02 19:41:32,244 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart
2023-03-02 19:41:32,755 - octoprint.server.api.system - WARNING - Command for core:restart failed with return code 1:
! 
! STDOUT:
! 
! 
! STDERR:
! sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper
! 
! sudo: a password is required

paukstelis, 2023-03-02:
I have a feeling that you restored some backup over the instance you created with the script, as neither of those commands will be used as part of octoprint_deploy.

reuter, 2023-03-02:
that could be. I did a settings import from my rpi server!!

reuter, 2023-03-02:
So what should these be then?

[image]

paukstelis, 2023-03-02:
> So what should these be then?
> 
> [image]

sudo systemctl restsart yourinstancename
sudo reboot
sudo shutdown now

reuter, 2023-03-02:
That did it!
Thank you!!

closed 2023-03-04 by paukstelis