paukstelis/octoprint_deploy #67: Camera No signal

dodolcool, 2023-03-06:
Hello
im running 4 instance with 4 generic webcam. the problem is only 2 camera that add the first time/ active first time will work fine and the 2 other will have no signal msg on the screen. i tried the cam on pc and they work fine. if i delete the instance that have working camera the other no signal camera will work and the one i delete and create again will have no signal msg. can you help?

dodolcool, 2023-03-06:
[image]
[image]

Here's what i get from check the No signal Cam

paukstelis, 2023-03-06:
You are running out of USB bandwidth. That is what the `No space left on device` means.
You should avoid plugging all your cameras into a USB hub. Spread them around your device. Running printers all off a hub is typically OK, but cameras use much more USB bandwidth.

dodolcool, 2023-03-06:
Is there any solution i only have 2 usb port. each running usb hub with 4 port. each hub running 2 machines and 2 webcam.

paukstelis, 2023-03-06:
No, not really. This is a limitation of the USB bus on your system.

dgrabla, 2024-07-01:
I've been researching this. If you tell the camera to use an uncompressed stream (YUY) you can use a kernel flag/quirk to fix the bandwith so not the theoretical maximum per camera is allocated. If you use compressed streams (MJPEG) you cannot use that, but there are kernel patches available you can use to tell the kernel to reserve a fix amount of bandwidth per camera (But no official way to do it with a vanilla kernel).   

The software solution is hacky and probably not something this script should try to do. Most people relocate the cameras on their USB ports to hit different USB controllers, and the ones that they STILL need more cameras they probably buy PCIe usb cards to add additional USB controllers. 

I think this info should be on the Readme.md as a warning. The whole point of the script is to install multiple instances, probably with at least one camera per instance so most users will hit this issue.