paukstelis/octoprint_deploy #70: One serial connection not working

S-A-M-J, 2023-03-14:
### Discussed in https://github.com/paukstelis/octoprint_deploy/discussions/69

<div type='discussions-op-text'>

<sup>Originally posted by **S-A-M-J** March 14, 2023</sup>
I've just used the script two create two instances. However one of them is unable to connect to the printer via serialport (receives no Answer). The setup worked flawlessly but after logging into the instance it is unable to connect. 

Printers: 2x Prusa mk3s+
Pi: Pi 4b
Os: Octopi

What i've tried: 
- Reinstalling the instance
- Reconnecting the printer
- Rebooting both pi and octoprint

</div>

S-A-M-J, 2023-03-14:
also tried switching the cables and ports. When switching cables only the same one instance works. When switching ports only the same one instance works. When only connecting the printer that is not connecting it still does not work. The log suggests that octoprint is sending data but receiving none.

S-A-M-J, 2023-03-14:
Ok new twist. The printer did connect once and messaged back but then declined to answer afterwards.

paukstelis, 2023-03-14:
Was just typing to check that when issue was closed. :)