paukstelis/octoprint_deploy #71: instance creation permission error

grantcoultrup, 2023-03-14:
Hi, I am encountering an error creating the first instance.

It looks like error code 13:
`PermissionError: [Errno 13] Permission denied: '/home/ampelOctopi1//.e3s1plus/config.yaml'`

Even though this error is received, it looks like the instance was still created (for example, I can "delete instance" and this instance is available to be deleted. Same for the associated USB camera.

When running Utility > Instance Status, I see that the instance failed.

Any known fix? Thanks!

[image]

paukstelis, 2023-03-14:
Can you verify that you have a config.yaml file in `~/.octoprint`? You could do this with: `ls ~/.octoprint/config.yaml`

paukstelis, 2023-03-14:
Further, you could verify the permissions in that directory: `ls -la ~/.octoprint`

grantcoultrup, 2023-03-14:
> Can you verify that you have a config.yaml file in `~/.octoprint`? You could do this with: `ls ~/.octoprint/config.yaml`

yep I've got config.yaml file.
[image]

paukstelis, 2023-03-14:
If you create a new instance, does that new directory also contain the config.yaml?
`ls -la ~/.e3s1plus` in your example

grantcoultrup, 2023-03-14:
> If you create a new instance, does that new directory also contain the config.yaml?
> `ls -la ~/.e3s1plus` in your example

it doesn't look like it... just in the root directory.
[image]

paukstelis, 2023-03-14:
The problem is that the config.yaml file is owned be `root`. It is not clear why that happened, as the script just copies everything from .octoprint (which doesn't have those permissions).

grantcoultrup, 2023-03-14:
> The problem is that the config.yaml file is owned be `root`. It is not clear why that happened, as the script just copies everything from .octoprint (which doesn't have those permissions).

got it... I'm not 100% proficient so this may be a dumb question... can I duplicate the .yaml file and then copy it to the instance directory? If so could you share some direct for that process?

paukstelis, 2023-03-14:
Doing a little searching it seems that there is a `sed` command in the script that can lead to random permission changes on some systems. It is not something I have observed before. I think there is fairly simple fix. I will implement in a new branch and have you test that out. Will get back shortly.

paukstelis, 2023-03-14:
Do the following:
`git -C octoprint_deploy pull`
`git -C octoprint_deploy checkout permission`
run script: `sudo octoprint_deploy/octoprint_deploy.sh`
delete the instance, then remake it

grantcoultrup, 2023-03-14:
Hm, still having issue... see image
[image]

paukstelis, 2023-03-14:
A different issue, however.

`sudo systemctl status instancename`

grantcoultrup, 2023-03-14:
> A different issue, however. `sudo systemctl status instancename`
> […](#)
[image]

grantcoultrup, 2023-03-14:
> A different issue, however. `sudo systemctl status instancename`
> […](#)

Still only seeing config.
[image]
yaml file in the root directory

paukstelis, 2023-03-14:
Sorry, ended up in a meeting. You need to give your instance name: `sudo sytemctl status e3s1plus`

paukstelis, 2023-03-15:
see what error you get from the status command.
Then try changing file ownership and restarting the instance:
`sudo chown ampelOctopi1 ~/.e3s1plus/config.yaml; sudo chgrp ampelOctopi1 ~/.e3s1plus/config.yaml`
`sudo systemctl restart e3s1plus`