paukstelis/octoprint_deploy #74: Camera Rework

onlywatanabe, 2023-04-01:
I'm working on OctoPi 1.0.0 with OctoPrint 1.8.7 build 20230328122604 from branch camera-streamer . It seems the way webcams work on the new OctoPi builds treats cameras differently. I had to use this new build in order to use raspicam/IMX219. With this new rebuild of camera streamer, it'd be nice to see if I could use this script to create two instances with two cameras. Or if anyone knows a method I could with this setup please let me know.

paukstelis, 2023-04-01:
I currently don't have any plans to work with the camera-streamer. My understanding is that you can use existing tools in the newest OctoPi builds to create camera services. All you would need to do is provide the appropriate URLs to your instances created with octoprint_deploy.

onlywatanabe, 2023-04-01:
Okay, I'll see how inquire how to do this in the new branch's github