paukstelis/octoprint_deploy #75: Webcam feed not showing in Obico

tekgecko, 2023-04-01:
I have succesfully deployed Octoprint and the webcam is working from the url: 

http://Octoprint.local:8002?action=stream

When linking the printer to Obico, it links but there is no image.

paukstelis, 2023-04-01:
If the camera is working, then perhaps this is a obico issue. You might try for support from them.

tekgecko, 2023-04-01:
I would have thought so too, but I have 3 other instances of Octoprint on RP3s that are connected to Obico with no issue.
It's just this one, since switching to Octoprint-deploy the image is no longer there..
I am wondering if it has something to do with the URL that Obico reads from?

tekgecko, 2023-04-01:
I just realised I had changed my Obico IP which was cuasing problems. Old cameras were still connecting somehow but the new ones couldn't reconfiguring the Django site solved this issue for anyone else having this problem.

closed 2023-04-01 by tekgecko