paukstelis/octoprint_deploy #78: RockyOS 8 Support

skl111, 2023-04-23:
Hello
I am requesting that Rocky OS 8 support be added!

skl111, 2023-04-23:
I got this working by adding the following
`dnf install epel-release
yum -y install python39 python39-devel git gcc-c++ policycoreutils-python-utils
`
and modifying the deploy script to not install python3 as that reverts python to default 3.6 version.

djjudas21, 2025-09-16:
I can also confirm that this was necessary on Rocky 9:

```
sudo dnf groupinstall "Development Tools"
sudo dnf install epel-release
sudo dnf install libbsd-devel-0.12.2-1.el9.x86_64
```

Probably could have got away with installing just a subset of the development tools rather than the whole group, but I was lazy.