paukstelis/octoprint_deploy #80: Why do you have to be in the directory level above the octoprint_deploy folder?

moltra, 2023-05-03:
I tried using the update function, but it said that the directory was not available.  

fatal: cannot change to 'octoprint_deploy': No such file or directory

I had already changed to the octoprint_deploy folder and started the script using the ./octoprint_deploy.sh command.

I do not understand why you have it setup this way.

paukstelis, 2023-05-03:
Simply because most people you use the script aren't that linux saavy. The README does say it assumes all commands come from the home directory.

moltra, 2023-05-04:
would it be hard to have it check to see what folder it is in?  and then change directory if needed?

paukstelis, 2023-05-05:
I have pushed a change that does this. Closing.