paukstelis/octoprint_deploy #82: instances not starting on reboot

EQNish, 2023-05-05:
after configuration, and several successful prints, Service\instances wont start.

need a LIST instances option in the menu

I can not find any instance of Octoprint other then the BASE template instance, what are the Instance names/directories, service names to individually start/stop daemon?

paukstelis, 2023-05-05:
There is. Use the instance status from the script menu.

EQNish, 2023-05-05:
hrmmm, then all my instances are gone... not sure I like this over the old method of cloning the octprint directory :)

paukstelis, 2023-05-05:
That is all the script does, but it clones the template instance. If you don't like it, don't use it. If your instances are gone then you deleted them, either through the script or manually. One of the common mistakes is people think they have made instances, but have answered `n` when it asks if they are ready to autodetect the printers. You must go through the autodetect to actually create the instance.

closed 2023-05-05 by paukstelis