paukstelis/octoprint_deploy #85: Octoeverywhere: All printers share same config

jpapiez, 2023-05-13:
How do we handle a plugin that requires unique configuration for each printer? In my case I'm trying to use OctoEverywhere and when trying to add the second printer it fails because the Octoeverywhere unique ID that's generated when the plugin is installed.

paukstelis, 2023-05-13:
The simplest thing to do  is to NOT install the plugin when you create your template instance (I am going to remove the cloud plugins in the future). Install it afterwards and each plugin will be configured separately.

For now, delete everything in the ~/.instancename/config.yaml related to octoeverywhere, or delete the plugin (and do clean up) the second instance. Restart. Then reinstall the plugin.

jpapiez, 2023-05-13:
Thank you Paul! I went the4 second route.. uninstall the plugin and all data, then reinstall for each instance.

paukstelis, 2023-05-13:
> Thank you Paul! I went the4 second route.. uninstall the plugin and all data, then reinstall for each instance. 

Fyi, no need to reinstall in both instances. Just reinstall in one and restart the other.