paukstelis/octoprint_deploy #87: [Debian] Cameras at the wrong port

Yippiehh, 2023-05-20:
Hello,

I have two printers running on Debian with octoprint_deploy (ustreamer, no haproxy) without any problems (with the printers).

I have installed the printers on port 5001 and 5002. The associated webcams on 7001 and 7002.

When I open the two printers via OctoPrint in the browser everything is correct and works perfectly.

But on both instances of the cameras are assigned to "http://OctoPrint.local:7003?action=stream".

So I always see only the image of one camera.

If I manually switch to 7001 or 7002, no video appears on OctoPrint. Is this a bug?

paukstelis, 2023-05-20:
Provide output from the following commands:
`cat /etc/systemd/system/cam_instancename1.service`
`cat /etc/systemd/system/cam_instancename2.sevice`

where instancename is replaced by the name of your instance.

Yippiehh, 2023-05-20:
Hallo, 

Port 7001: (cat /etc/systemd/system/cam_Mega.service)
[Unit]
Description=the OctoPi(buntu) ustreamer daemon with the user specified config
After=network.online.target
Wants=network.online.target

[Service]
User=pi
ExecStart=/home/pi/ustreamer/ustreamer -d /dev/cam_Mega -s 0.0.0.0 -m MJPEG -r 640x480 -f 5 -p 7002 --device-timeout 8 --device-error-delay 8

[Install]
WantedBy=multi-user.target

Port 7002: cat /etc/systemd/system/cam_Vyper.service
[Unit]
Description=the OctoPi(buntu) ustreamer daemon with the user specified config
After=network.online.target
Wants=network.online.target

[Service]
User=pi
ExecStart=/home/pi/ustreamer/ustreamer -d /dev/cam_Vyper -s 0.0.0.0 -m MJPEG -r 640x480 -f 5 -p 7003 --device-timeout 8 --device-error-delay 8

[Install]
WantedBy=multi-user.target

Both Ports are one number higher then expected (and used in installation). 

But ….

…. after a clean Debian reinstall and octoprint_deply install, I installed the cams on 7000 and 7001 and now they are correct in both *.service files with 7001 and 7002. 

And the best: http://OctoPrint.local:7001?action=stream <http://octoprint.local:7001/?action=stream> and http://OctoPrint.local:7002?action=stream <http://octoprint.local:7002/?action=stream> are running with the correct cameras! :-)

So the problem with two ports showing the same camera must be my failure, but the increased number is a failure of octoprint_deploy. Correct?

cu, Alex Gebauer

paukstelis, 2023-05-20:
I'll have to look closer. It isn't an issue I have seen before, but forcing ports to be greater than 7000 is a feature I added fairly recently.

paukstelis, 2023-05-21:
yeah, this was just poor planning on my part. I will be fixing the issue shortly.

paukstelis, 2023-05-22:
completed. https://github.com/paukstelis/octoprint_deploy/commit/aea4c30e3e83dba253a7d8320f540d6e15d78fb0

closed 2023-05-22 by paukstelis