paukstelis/octoprint_deploy #90: missing commnad uuidgen raspberry

KyferEz, 2023-06-01:
Getting this error when setting up first printer using Raspberry Pi OS:
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian

[image]

[image]

Found this which seems related: https://forums.raspberrypi.com/viewtopic.php?t=268239

paukstelis, 2023-06-01:
Unfortunately, they have given it a different name. Try: `sudo apt install uuid-runtime`
It is a non-critical error, so your instance should be fine, unless you want to reinstall it.

KyferEz, 2023-06-02:
Paul, thank you, that worked!

closed 2023-06-02 by KyferEz