paukstelis/octoprint_deploy #93: Python and Pip not working

RVillan0, 2023-06-14:
Hi. I am using the script to download octoprint on ubuntu. I have updated and upgraded the terminal. Go to run the script for multiple instances, select 1) Prepare system > 2) Ubuntu > y for begin then it goes till I get this message 

Ready to begin? [y/n]: y
Adding current user to dialout and video groups.
Adding systemctl and reboot to sudo
This will install necessary packages, download and install OctoPrint and setup a template instance on this machine.
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.9ubuntu3).
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
make is already the newest version (4.3-4.1build1).
libbsd-dev is already the newest version (0.11.5-1).
libevent-dev is already the newest version (2.1.12-stable-1build3).
libffi-dev is already the newest version (3.4.2-4).
libjpeg-dev is already the newest version (8c-2ubuntu10).
libjpeg8-dev is already the newest version (8c-2ubuntu10).
libyaml-dev is already the newest version (0.2.2-1build2).
python-is-python3 is already the newest version (3.9.2-2).
uuid-runtime is already the newest version (2.37.2-4ubuntu3).
python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.1).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
haproxy is already the newest version (2.4.22-0ubuntu0.22.04.1).
python3-dev is already the newest version (3.10.6-1~22.04).
ssh is already the newest version (1:8.9p1-3ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
Package 'brltty' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Enabling ssh server...
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable ssh
Installing OctoPrint virtual environment in /home/print/OctoPrint
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt install python3.10-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/print/OctoPrint/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

sudo: /home/print/OctoPrint/bin/pip: command not found
sudo: /home/print/OctoPrint/bin/pip: command not found
sudo: /home/print/OctoPrint/bin/pip: command not found
WARNING! WARNING! WARNING!
OctoPrint has not been installed correctly.
Please answer Y to remove everything and try running prepare system again.
Remove everything? [y/n]: 

I try the apt install python3.10-venv command and I get

sudo apt install python3.10-venv 
[sudo] password for print: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python3.10-venv is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python3.10-venv' has no installation candidate

I'm pretty sure this is because of pip. Not sure why it is not downloading. So i tried this from the octoprint website:

sudo apt install python3-pip python3-dev python3-setuptools python3-venv git libyaml-dev build-essential
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python3-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python3-venv is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python3-pip' has no installation candidate
E: Package 'python3-venv' has no installation candidate

Do you have any idea what is causing this?

paukstelis, 2023-06-14:
What version of Ubuntu are you using and what hardware have you installed it on?

RVillan0, 2023-06-14:
Version of Ubuntu is Ubuntu 22.04.2 LTS and the hardware I am not sure on. The computer is a Dell Optiplex 9020

paukstelis, 2023-06-14:
What do you get with `python3 --version`?

RVillan0, 2023-06-14:
Python 3.10.6

paukstelis, 2023-06-14:
There is something wrong with how the packages are configured. Those are all default packages that are available in 22.04.
Try a `sudo apt update; sudo apt upgrade`

RVillan0, 2023-06-14:
Im not sure what else to do

paukstelis, 2023-06-14:
I would reinstall Ubuntu. There is clearly something wrong, as those packages should certainly be available in Ubuntu jammy.
[image]

RVillan0, 2023-06-14:
Yeah i'm going to try that out

paukstelis, 2023-06-15:
That's just not accurate. My firsts tests are always on a VM running 22.04-server with no modifications or updates.

nncrypted, 2023-06-15:
So don't update is what you're saying.

Mobi97213, 2023-07-17:
I have installed Fedora-KDE-38-1.6.aarch64 on my raspberry pi 4B.
My python version is: 3.11.4
There is no Python3.10-venv (or Python3.11-venv) in the repositories.
There is python3-pylint-venv and python3-pytest-venv. I installed those.
Running octoprint_deploy 1.0.3
Selected 1-Prepare
Selected 2-Fedora
Responded Y to Disable SELinux prompt.
Message: failed to enable unit ssh.service, does not exist (openssh fully installed). ignored since I don't need ssh.
The script does not seem to check for Cython as a dependency.
python3-Cython-0.29.33-1.fc38.aarch34 was not installed. Installed with dnf.
Apparently, there is an error reported on line 201, get_source_files.
Message: AttributeError: cython_sources
Installation ends with error: Getting requirements to build wheel: did not run successfully

I'm stuck. Any ideas on what I can do next?

paukstelis, 2023-07-17:
> I have installed Fedora-KDE-38-1.6.aarch64 on my raspberry pi 4B. My python version is: 3.11.4 There is no Python3.10-venv (or Python3.11-venv) in the repositories. There is python3-pylint-venv and [...]

It would be useful to see which packages fail when you go give the following command:
```
sudo  dnf install -y \
    gcc \
    python3-devel \
    cmake \
    libjpeg-turbo-devel \
    libbsd-devel \
    libevent-devel \
    haproxy \
    openssh \
    openssh-server \
    libffi-devel \
    libcamera-devel \
    v4l-utils \
    xxd \
    openssl-devel \
    rsync
```

Mobi97213, 2023-07-17:
Last metadata expiration check: 0:41:07 ago on Mon 17 Jul 2023 11:59:09 AM PDT.
Package gcc-13.1.1-4.fc38.aarch64 is already installed.
Package python3-devel-3.11.4-1.fc38.aarch64 is already installed.
Package cmake-3.26.4-4.fc38.aarch64 is already installed.
Package libjpeg-turbo-devel-2.1.4-2.fc38.aarch64 is already installed.
Package libbsd-devel-0.11.7-4.fc38.aarch64 is already installed.
Package libevent-devel-2.1.12-8.fc38.aarch64 is already installed.
Package haproxy-2.7.8-1.fc38.aarch64 is already installed.
Package openssh-9.0p1-15.fc38.aarch64 is already installed.
Package openssh-server-9.0p1-15.fc38.aarch64 is already installed.
Package libffi-devel-3.4.4-2.fc38.aarch64 is already installed.
Package libcamera-devel-0.0.4-3.fc38.aarch64 is already installed.
Package v4l-utils-1.22.1-5.fc38.aarch64 is already installed.
Package xxd-2:9.0.1671-1.fc38.aarch64 is already installed.
Package openssl-devel-1:3.0.9-1.fc38.aarch64 is already installed.
Package rsync-3.2.7-2.fc38.aarch64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

paukstelis, 2023-07-17:
I think I know the issue. What do you get with `sudo dnf install python3.10`?

paukstelis, 2023-07-18:
https://github.com/yaml/pyyaml/issues/724

This appears to be the issue that is impacting installs.

paukstelis, 2023-07-18:
It is fixed as of today. Start with system prepare again and you should be good to go.

closed 2023-07-18 by paukstelis

Mobi97213, 2023-07-18:
> It is fixed as of today. Start with system prepare again and you should be good to go. https://community.octoprint.org/t/new-release-1-9-2/52970

SUCCESS! Thank you :+1: