paukstelis/octoprint_deploy #95: Cant't get instances working

botacini, 2023-06-29:
Hi, first of all thank you for the awesome toll, it helped me for a long time before.
I used to have two instances working just fine, but for some reason I've never been able to add cameras to them, they weren't extremely necessary at the time so I leaved as it was.  
I was running an out of date version of both octoprint_deploy and the octoprint itself (and I've tried to update the octoprint via the browser UI several times unsuccessfully). 
So, to try to solve all of that I saved backups of all two instances,  uninstalled everything and upgraded the octoprint deploy. I've got some haproxy issues in the way but managed to get it running too.
Long history short, I'm now trying to create the instances again but the instances doesn't start as they are created, so I ran the diagnostic toll and this is the output: 

> Select an option: 8
> octoprint_deploy diagnostic information. Please provide ALL output for support help
> **************************************
> /etc/octoprint_deploy
> **************************************
> haproxy: true
> type: linux
> streamer: none
> **************************************
> /etc/octoprint_instances
> **************************************
> instance:Impressora1 port:5000 udev:true
> **************************************
> /etc/octoprint_cameras
> **************************************
> camera:cam_Impressora1 port:8001 udev:true
> **************************************
> /etc/udev/rules.d/99-octoprint.rules
> **************************************
> SUBSYSTEM=="video4linux",KERNELS=="5-1.1:1.0", SUBSYSTEMS=="usb", ATTR{index}=="0", DRIVERS=="uvcvideo", SYMLINK+="cam_Impressora_1"
> SUBSYSTEM=="tty", ATTRS{serial}=="AK064KRY", SYMLINK+="octo_Impressora1"
> SUBSYSTEM=="video4linux",KERNELS=="4-1.4:1.0", SUBSYSTEMS=="usb", ATTR{index}=="0", DRIVERS=="uvcvideo", SYMLINK+="cam_Impressora1"
> **************************************
> ● Impressora1.service - The snappy web interface for your 3D printer
>      Loaded: loaded (/etc/systemd/system/Impressora1.service; enabled; vendor preset: enabled)
>      Active: failed (Result: exit-code) since Thu 2023-06-29 17:40:20 -03; 1min 21s ago
>     Process: 550 ExecStart=serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT} (code=exited, status=203/EXEC)
>    Main PID: 550 (code=exited, status=203/EXEC)
>         CPU: 9ms
> 
> jun 29 17:40:20 raspberry systemd[1]: Started The snappy web interface for your 3D printer.
> jun 29 17:40:20 raspberry systemd[550]: Impressora1.service: Failed to locate executable serve: No such file or directory
> jun 29 17:40:20 raspberry systemd[550]: Impressora1.service: Failed at step EXEC spawning serve: No such file or directory
> jun 29 17:40:20 raspberry systemd[1]: Impressora1.service: Main process exited, code=exited, status=203/EXEC
> jun 29 17:40:20 raspberry systemd[1]: Impressora1.service: Failed with result 'exit-code'.
> sudo: opção não reconhecida “--basedir=/home/sgtbotacini/.”
> usage: sudo -h | -K | -k | -V
> usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
> usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user]
>             [command]
> usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group]
>             [-h host] [-p prompt] [-R directory] [-T timeout] [-u user]
>             [VAR=value] [-i|-s] [<command>]
> usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group]
>             [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
> sudo: opção não reconhecida “--basedir=/home/sgtbotacini/.”
> usage: sudo -h | -K | -k | -V
> usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
> usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user]
>             [command]
> usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group]
>             [-h host] [-p prompt] [-R directory] [-T timeout] [-u user]
>             [VAR=value] [-i|-s] [<command>]
> usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group]
>             [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
> **************************************
> Unit cam_Impressora1.service could not be found.
> **************************************
> ● haproxy.service - HAProxy Load Balancer
>      Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
>      Active: active (running) since Thu 2023-06-29 17:40:30 -03; 1min 12s ago
>        Docs: man:haproxy(1)
>              file:/usr/share/doc/haproxy/configuration.txt.gz
>     Process: 678 ExecStartPre=/usr/sbin/haproxy -f $CONFIG -c -q $EXTRAOPTS (code=exited, status=0/SUCCESS)
>    Main PID: 952 (haproxy)
>       Tasks: 5 (limit: 2259)
>      Memory: 36.8M
>         CPU: 375ms
>      CGroup: /system.slice/haproxy.service
>              ├─952 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
>              └─957 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
> 
> jun 29 17:40:27 raspberry systemd[1]: Starting HAProxy Load Balancer...
> jun 29 17:40:30 raspberry haproxy[952]: Proxy Impressora1 started.
> jun 29 17:40:30 raspberry haproxy[952]: Proxy Impressora1 started.
> jun 29 17:40:30 raspberry systemd[1]: Started HAProxy Load Balancer.
> jun 29 17:40:30 raspberry haproxy[952]: [NOTICE] 179/174029 (952) : New worker #1 (957) forked
> jun 29 17:40:30 raspberry haproxy[952]: Proxy cam_Impressora1 started.
> jun 29 17:40:30 raspberry haproxy[952]: Proxy cam_Impressora1 started.
> 

Thank you again and sorry for any ridiculous mistake I may have made.

paukstelis, 2023-06-29:
OctoPrint is not installed on your system. 
You should use the remove argument and start again (sudo octoprint_deploy/octoprint_deploy.sh remove). If you have backups of your instances you can restore those AFTER you have recreated them with octoprint_deploy

botacini, 2023-07-01:
Hi, thanks for the reply!
I've managed to execute the REMOVE command and after a couple errors I realized that in order to get through the "prepare the system" with the Octoprint_deploy without any errors in the process, I needed to install the haproxy manually before it. However, even it hadn't show any error messages during the preparation process, when I choose "yes" for using Haproxy it was unable to start the service and eventually I gave up of trying to get the haproxy to work, there's no problem for me to add the port to the address in the browser and at the end I've managed to create two instances.
However, for some reason I cant add a camera to each of them. Every time I try to add a new camera having a first one already working the second one doesn't work and I get the following error message:

> 
> octoprint_deploy 1.0.1
> *************************
> 
> 1) Add instance	    3) Add Camera	5) Utilities	    7) Update
> 2) Delete instance  4) Delete Camera	6) Backup Menu	    8) Quit
> Select operation: 3
> Adding camera
> 1) Impressora1
> 2) Impressora2
> 3) Quit
> Select instance number to add camera to: 2
> Selected instance for camera: Impressora2
> It appears this instance already has at least one camera.
> Do you want to add an additional camera for this instance? [y/n]: y
> Enter a number for this camera.
> Ex. entering 2 will setup a service called cam2_Impressora2
> 
> 2
> 
> 
> Verify the camera is currently unplugged from USB.....
> Is the camera you are trying to detect unplugged from USB? [y/n]: n
> Plug your camera in via USB now (detection time-out in 1 min)
> /dev/v4l/by-id/usb-Generic_USB2.0_PC_CAMERA-video-index0
> Camera Serial Number not detected
> Camera will be setup with physical USB address of 1-1:1.0.
> The camera will have to stay plugged into this location.
> /home/sgtbotacini/octoprint_deploy/cameras.sh: linha 227: 8001 udev:true: erro de sintaxe na expressão (token de erro é "udev:true")

Doesn't matter if I try adding the second camera to the second instance or as an additional camera for the first one, it always end in this given error. Both cameras work, but only if it is the first to be added.

Thank you very much for your support!

paukstelis, 2023-07-01:
Thank you for reporting this. Yes, it is a bug. I will have it fixed shortly.

paukstelis, 2023-07-01:
That error is now fixed. Update octoprint_deploy and try adding second camera again. Closing.

closed 2023-07-01 by paukstelis

botacini, 2023-07-01:
I've tried to use the update command inside the application and I get this message:

error: cannot open .git/FETCH_HEAD: Permissão negada

Thanks for your support again.

paukstelis, 2023-07-01:
I am not sure why you get that. You can always use the command line: `git -C octoprint_deploy pull`

botacini, 2023-07-01:
Well, I worked it around by removing the Octoprint_deploy folder and cloning it again, it kinda worked, now I'm able to add both cams, however for some reason they appear to be hosted in the same port 8001 and now I'm getting the printer one image in both instances. Diagnostic information below:

> octoprint_deploy diagnostic information. Please provide ALL output for support help
> **************************************
> /etc/octoprint_deploy
> **************************************
> octoexec: /home/sgtbotacini/OctoPrint/bin/octoprint
> octopip: /home/sgtbotacini/OctoPrint/bin/pip
> streamer: none
> **************************************
> /etc/octoprint_instances
> **************************************
> instance:Impressora2 port:5001 udev:true
> instance:Impressora1 port:5002 udev:true
> **************************************
> /etc/octoprint_cameras
> **************************************
> camera:cam_Impressora1 port:8001 udev:true
> camera:cam_Impressora2 port:8001 udev:true
> **************************************
> /etc/udev/rules.d/99-octoprint.rules
> **************************************
> KERNELS=="3-1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Impressora2"
> SUBSYSTEM=="tty", ATTRS{serial}=="AK064KRY", SYMLINK+="octo_Impressora1"
> SUBSYSTEM=="video4linux",KERNELS=="5-1.4:1.0", SUBSYSTEMS=="usb", ATTR{index}=="0", DRIVERS=="uvcvideo", SYMLINK+="cam_Impressora1"
> SUBSYSTEM=="video4linux",KERNELS=="1-8:1.0", SUBSYSTEMS=="usb", ATTR{index}=="0", DRIVERS=="uvcvideo", SYMLINK+="cam_Impressora2"
> lrwxrwxrwx 1 root root 7 Jul  1 19:54 /dev/octo_Impressora1 -> ttyUSB0
> lrwxrwxrwx 1 root root 6 Jul  1 19:54 /dev/cam_Impressora1 -> video0
> lrwxrwxrwx 1 root root 6 Jul  1 19:54 /dev/cam_Impressora2 -> video2
> **************************************
> * Impressora2.service - The snappy web interface for your 3D printer
>      Loaded: loaded (/etc/systemd/system/Impressora2.service; enabled; vendor preset: enabled)
>      Active: active (running) since Sat 2023-07-01 19:54:31 -03; 5min ago
>    Main PID: 2881 (octoprint)
>       Tasks: 27 (limit: 2259)
>      Memory: 82.6M
>         CPU: 1min 16.333s
>      CGroup: /system.slice/Impressora2.service
>              `-2881 /home/sgtbotacini/OctoPrint/bin/python3 /home/sgtbotacini/OctoPrint/bin/octoprint serve --config=/home/sgtbotacini/.Impressora2/config.yaml --basedir=/home/sgtbotacini/.Impressora2 --port=5001
> 
> Jul 01 19:58:36 raspberry octoprint[2881]: | UltimakerFormat=missing (None)
> Jul 01 19:58:36 raspberry octoprint[2881]: | PrusaSlicerThumbnail=missing (None)
> Jul 01 19:58:36 raspberry octoprint[2881]: | costestimation=enabled (3.5.0)
> Jul 01 19:58:36 raspberry octoprint[2881]: 
> Jul 01 19:58:36 raspberry octoprint[2881]: 2023-07-01 19:58:36,141 - octoprint.plugins.PrintJobHistory - WARNING - Filamentracking is disabled, but some plugins are installed!
> Jul 01 19:58:36 raspberry octoprint[2881]: 2023-07-01 19:58:36,145 - octoprint.server.util.flask - INFO - Passively logging in user botacini from 192.168.15.18
> Jul 01 19:58:36 raspberry octoprint[2881]: 2023-07-01 19:58:36,146 - octoprint.access.users - INFO - Logged in user: botacini
> Jul 01 19:58:40 raspberry octoprint[2881]: 2023-07-01 19:58:40,284 - octoprint.server.util.sockjs - INFO - User botacini logged in on the socket from client 192.168.15.18
> Jul 01 19:58:42 raspberry octoprint[2881]: 2023-07-01 19:58:42,225 - octoprint.plugins.tracking - INFO - Sent tracking event webui_load, payload: {'browser_name': 'Chrome', 'browser_version': '114.0.0.0', 'os_name': 'Windows', 'os_version': '10'}
> Jul 01 19:59:43 raspberry octoprint[2881]: 2023-07-01 19:59:43,609 - octoprint.plugins.telegram.listener - ERROR - Setting status: Telegram API responded with code 409. Waiting 2 minutes before trying again.
> None
> {'bitrate': '10000k',
>  'cleanTmpAfterDays': 7,
>  'defaultWebcam': 'classic',
>  'ffmpeg': None,
>  'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
>                       '{videocodec} -threads {threads} -b:v {bitrate} -f '
>                       '{containerformat} -y {filters} "{output}"',
>  'ffmpegThreads': 1,
>  'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
>                                '0.7 "{output}"',
>  'ffmpegVideoCodec': 'libx264',
>  'snapshotWebcam': 'classic',
>  'timelapse': {'fps': 25,
>                'options': {'capturePostRoll': None,
>                            'interval': None,
>                            'retractionZHop': None},
>                'postRoll': 0,
>                'type': 'off'},
>  'timelapseEnabled': True,
[... 58 lines cut ...]
> Jul 01 19:46:58 raspberry ustreamer[2369]: -- INFO  [435.169    stream] -- Capturing ...
> Jul 01 19:48:19 raspberry ustreamer[2369]: -- INFO  [516.598      http] -- HTTP: NEW client (now=1): [192.168.15.18]:50060, id=1e52bf7936f9cfa6
> Jul 01 19:49:52 raspberry ustreamer[2369]: -- INFO  [609.605      http] -- HTTP: NEW client (now=2): [192.168.15.18]:50194, id=389d44664103be00
> Jul 01 19:54:52 raspberry ustreamer[2369]: -- INFO  [909.201      http] -- HTTP: DEL client (now=1): [192.168.15.18]:50194, id=389d44664103be00, Resource temporarily unavailable (reading,eof)
> Jul 01 19:55:31 raspberry ustreamer[2369]: -- INFO  [948.509      http] -- HTTP: NEW client (now=2): [192.168.15.18]:50631, id=eefc8e1e71473e01
[... cut ...]
> Jul 01 19:40:32 raspberry ustreamer[559]: -- INFO  [40.715      main] -- Using internal blank placeholder
> Jul 01 19:40:32 raspberry ustreamer[559]: -- ERROR [40.716      main] -- Can't bind HTTP on [0.0.0.0]:8001: Address already in use
> Jul 01 19:40:23 raspberry systemd[1]: cam_Impressora2.service: Main process exited, code=exited, status=1/FAILURE
> Jul 01 19:40:23 raspberry systemd[1]: cam_Impressora2.service: Failed with result 'exit-code'.
> **************************************
> Unit haproxy.service could not be found.

Thank you for your support!

paukstelis, 2023-07-02:
Yes, still another bug on my part. I will get it fixed soon.

paukstelis, 2023-07-02:
Thank you for your patience. It should now be fixed.