paukstelis/octoprint_deploy #103: possible to install/run as non-login service user

zerog2k, 2023-07-16:
currently, it seems it wants to run only as a user, and only installed from sudo with that user
resulting octoprint is running with this non-root user
but this requires it to use a real user which has login/shell privilege. If possible I'd rather run octoprint with an unprivileged user which cannot login (e.g. shell = /sbin/nologin, etc)
I wonder if there is a way to run the script as root, but specify either some service user to use, or create, e.g. `octoprint`

paukstelis, 2023-07-17:
No plans to include that. A user makes things more predictable with respective to paths where things are installed.

fake-name, 2023-08-10:
Uh, none of the things you list are precluded by a non-login user.

paukstelis, 2023-08-10:
Even if it is somewhere weird, the shell knows where it is. That's the point of $HOME is a variable.