paukstelis/octoprint_deploy #108: Problem connecting to Ender 3 printers

joelstitch, 2023-07-23:
So After a ton of hours wasted trying to get this to work I decided to once again start from scratch.

1. Installed raspbian.
2. ssh into the pi, update, install git, install octoprint_deploy.
3. run octoprint_deploy
4. Prepare the system
5. Add Instance
6. open instance web address.
7. make sure its showing the right Serial Port
8. Click Connect.
9. get error
**State: Offline after error
No more candidates to test, and no working port/baudrate combination detected.**

Please help me understand what I'm doing wrong so I can get this to work.why does it detects the printer when adding the instance but refuses to connect on Octoprint?

joelstitch, 2023-07-23:
octoprint_deploy diagnostic information. Please provide ALL output for support h                                                                                                                     elp
**************************************
/etc/octoprint_deploy
**************************************
haproxy: true
streamer: ustreamer
octoexec: /home/pi/OctoPrint/bin/octoprint
octopip: /home/pi/OctoPrint/bin/pip
**************************************
/etc/octoprint_instances
**************************************
instance:ender3 port:5000 udev:true
instance:ender3pro port:5001 udev:true
cat: /etc/octoprint_cameras**************************************
/etc/octoprint_cameras
**************************************
: No such file or directory
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
KERNELS=="1-1.5:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_ender3"
KERNELS=="1-1.4:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_ender3pro"
lrwxrwxrwx 1 root root 7 Jul 23 16:54 /dev/octo_ender3 -> ttyUSB0
ls: cannot access '/dev/cam*': No such file or directory
**************************************
● ender3.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/ender3.service; enabled; vendor preset:                                                                                                                      enabled)
     Active: active (running) since Sun 2023-07-23 16:54:41 CDT; 23min ago
   Main PID: 386 (octoprint)
      Tasks: 22 (limit: 1595)
        CPU: 1min 8.516s
     CGroup: /system.slice/ender3.service
             └─386 /home/pi/OctoPrint/bin/python3 /home/pi/OctoPrint/bin/octopri                                                                                                                     nt serve --config=/home/pi/.ender3/config.yaml --basedir=/home/pi/.ender3 --port                                                                                                                     =5000

Jul 23 17:15:34 raspberrypi octoprint[386]: 2023-07-23 17:15:34,099 - octoprint.                                                                                                                     plugins.action_command_notification - INFO - Notifications cleared
Jul 23 17:15:39 raspberrypi octoprint[386]: 2023-07-23 17:15:38,984 - octoprint.                                                                                                                     util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial c                                                                                                                     onnection"
Jul 23 17:15:39 raspberrypi octoprint[386]: 2023-07-23 17:15:38,993 - octoprint.                                                                                                                     util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
Jul 23 17:15:39 raspberrypi octoprint[386]: 2023-07-23 17:15:39,029 - octoprint.                                                                                                                     util.comm - INFO - Changing monitoring state from "Opening serial connection" to                                                                                                                      "Connecting"
Jul 23 17:15:39 raspberrypi octoprint[386]: 2023-07-23 17:15:39,042 - octoprint.                                                                                                                     util.comm - INFO - M110 detected, setting current line number to 0
Jul 23 17:15:39 raspberrypi octoprint[386]: 2023-07-23 17:15:39,196 - octoprint.                                                                                                                     server - INFO - Autorefresh of serial port list stopped
Jul 23 17:16:23 raspberrypi octoprint[386]: 2023-07-23 17:16:23,534 - octoprint.                                                                                                                     util.comm - INFO - M110 detected, setting current line number to 0
Jul 23 17:16:38 raspberrypi octoprint[386]: 2023-07-23 17:16:38,542 - octoprint.                                                                                                                     util.comm - INFO - Changing monitoring state from "Connecting" to "Offline"
Jul 23 17:16:38 raspberrypi octoprint[386]: 2023-07-23 17:16:38,564 - octoprint.                                                                                                                     server - INFO - Starting autorefresh of serial port list
Jul 23 17:16:38 raspberrypi octoprint[386]: 2023-07-23 17:16:38,573 - octoprint.                                                                                                                     plugins.action_command_notification - INFO - Notifications cleared
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
[... cut ...]
             └─654 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

Jul 23 16:54:49 raspberrypi systemd[1]: Starting HAProxy Load Balancer...
Jul 23 16:54:49 raspberrypi haproxy[652]: [WARNING] 203/165449 (652) : parsing [/etc/haproxy/haproxy.cfg:38] : a 'http-request' rule placed after a 'use_backend' rule will still be processed before.
Jul 23 16:54:49 raspberrypi haproxy[652]: [NOTICE] 203/165449 (652) : New worker #1 (654) forked
Jul 23 16:54:49 raspberrypi systemd[1]: Started HAProxy Load Balancer.

paukstelis, 2023-07-23:
Based on the previous issues you opened and this one I would say that you likely installed Klipper firmware on your printers. Once you do that, they will no longer connect to OctoPrint as normal USB devices. They now need to connect to OctoPrint through Klipper. Re-install Marlin firmware and you can see if it works.

The printers are still detected during the setup because what is actually being detected is the USB->UART chip.

closed 2023-07-26 by paukstelis