paukstelis/octoprint_deploy #113: Cannot install octoprint without printer connected

fake-name, 2023-08-10:
Functionally the title. 

I'm trying to get octoprint working on a SBC so I can validate some additional hardware accessories. I do not have the printer connected to the SBC at the moment. 

Apparently you cannot deploy octoprint with this script without having it physically connected to the printer.

Anyways, I commented out a bunch of exit conditions in `instance.sh` and it finished fine, even with no printer present, and I now have octoprint happily running. I can now move on to dealing with getting some auxiliary relays to play nice with octoprint, before installing the entire hardware assembly into my printer's controller enclosure.

paukstelis, 2023-08-10:
> Apparently you cannot deploy octoprint with this script without having it physically connected to the printer.

This is incorrect. You can install OctoPrint and create a first instance without having a printer physically connected.

fake-name, 2023-08-10:
If so, how?

If you answer no to the autodetection step, it kicks you back to the main menu. If it doesn't detect a usb-serial device, it also kicks you back to the main menu.

paukstelis, 2023-08-10:
It works for the first instance created during preparation (if you want to start over: sudo octoprint_deploy/octoprint_deploy.sh remove). You start the auto-detection, let it time out. Then Y when ready to proceed.

[image]

closed 2023-12-20 by paukstelis