paukstelis/octoprint_deploy #114: Second Camera

dillario, 2023-08-13:
Hello,
[image]

I made 2 instances for my two printers. But I can't get the second camera to work. Can someone tell me what to enter here? Thanks.

paukstelis, 2023-08-13:
octoprint_deploy will automatically fill that, depending on if you used haproxy for cameras.

Please provide the diagnostic output from the Utilities menu.

dillario, 2023-08-13:
Thanks. Don't quite understand what I have to do. Can you explain it to me in more Details?

paukstelis, 2023-08-13:
`sudo octoprint_deploy/octoprint_deploy.sh`
Select `Utilities`
Select `Diagnostic Output`
Copy and paste the output here.

dillario, 2023-08-13:
octoprint_deploy diagnostic information. Please provide ALL output for support h                                elp
**************************************
/etc/octoprint_deploy
**************************************
octoexec: /home/pi/oprint/bin/octoprint
octopip: /home/pi/oprint/bin/pip
haproxy: true
octopi: true
streamer: camera-streamer
**************************************
/etc/octoprint_instances
**************************************
instance:octoprint port:5000 udev:true
instance:MegaS port:5001 udev:true
instance:KobraMax port:5002 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_MegaS port:8001 udev:true
camera:cam_KobraMax port:8002 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
SUBSYSTEM=="tty", ATTRS{serial}=="0001", SYMLINK+="octo_octoprint"
SUBSYSTEM=="tty", ATTRS{serial}=="0001", SYMLINK+="octo_MegaS"
SUBSYSTEM=="video4linux", ATTRS{serial}=="9A42DA90", ATTR{index}=="0", SYMLINK+=                                "cam_MegaS"
KERNELS=="1-1.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_KobraMax"
SUBSYSTEM=="video4linux", ATTRS{serial}=="69C4E050", ATTR{index}=="0", SYMLINK+=                                "cam_KobraMax"
lrwxrwxrwx 1 root root 7 Aug 13 09:21 /dev/octo_KobraMax -> ttyUSB1
lrwxrwxrwx 1 root root 7 Aug 13 09:21 /dev/octo_MegaS -> ttyUSB0
lrwxrwxrwx 1 root root 7 Aug 13 09:21 /dev/octo_octoprint -> ttyUSB0
lrwxrwxrwx 1 root root 6 Aug 13 09:21 /dev/cam2_KobraMax -> video2
lrwxrwxrwx 1 root root 6 Aug 13 09:21 /dev/cam_KobraMax -> video2
lrwxrwxrwx 1 root root 6 Aug 13 09:21 /dev/cam_MegaS -> video0
**************************************
● octoprint.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor pres                                et: enabled)
     Active: active (running) since Sun 2023-08-13 08:59:27 BST; 4h 44min ago
   Main PID: 2632 (octoprint)
      Tasks: 14 (limit: 3135)
        CPU: 3min 29.772s
     CGroup: /system.slice/octoprint.service
             └─2632 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint ser                                ve --host=127.0.0.1 --port=5000

Aug 13 11:14:30 octopi octoprint[2632]: 2023-08-13 11:14:30,789 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 11:29:30 octopi octoprint[2632]: 2023-08-13 11:29:30,790 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 11:44:30 octopi octoprint[2632]: 2023-08-13 11:44:30,791 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 11:59:30 octopi octoprint[2632]: 2023-08-13 11:59:30,792 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 12:14:30 octopi octoprint[2632]: 2023-08-13 12:14:30,793 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 12:29:30 octopi octoprint[2632]: 2023-08-13 12:29:30,795 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 12:44:30 octopi octoprint[2632]: 2023-08-13 12:44:30,796 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 12:59:30 octopi octoprint[2632]: 2023-08-13 12:59:30,797 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 13:14:30 octopi octoprint[2632]: 2023-08-13 13:14:30,798 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
Aug 13 13:29:30 octopi octoprint[2632]: 2023-08-13 13:29:30,800 - octoprint.serv                                er.heartbeat - INFO - Server heartbeat <3
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
               'options': {'capturePostRoll': None,
[... 96 lines cut ...]
Unit cam_KobraMax.service could not be found.
**************************************
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-08-13 09:25:29 BST; 4h 19min ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
    Process: 5504 ExecStartPre=/usr/sbin/haproxy -f $CONFIG -c -q $EXTRAOPTS (code=exited, status=0/SUCCESS)
   Main PID: 5506 (haproxy)
      Tasks: 5 (limit: 3135)
        CPU: 46.690s
     CGroup: /system.slice/haproxy.service
             ├─5506 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
             └─5508 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

[... cut ...]
Aug 13 13:14:44 octopi haproxy[5508]: ::ffff:192.168.178.39:45358 [13/Aug/2023:13:14:44.125] public octoprint/octoprint1 0/0/0/26/26 200 937 - - ---- 1/1/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Aug 13 13:35:26 octopi haproxy[5508]: ::ffff:192.168.178.30:36490 [13/Aug/2023:13:35:26.318] public octoprint/octoprint1 0/0/0/28/28 200 937 - - ---- 1/1/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Aug 13 13:36:21 octopi haproxy[5508]: ::ffff:192.168.178.42:42627 [13/Aug/2023:13:36:21.204] public octoprint/octoprint1 0/0/1/16/17 200 937 - - ---- 1/1/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"

paukstelis, 2023-08-13:
You are using OctoPi, so setting up cameras must be done manually, not with octoprint_deploy (unless you change the streamer to ustreamer).

paukstelis, 2023-08-13:
octoprint_deploy will do it for you

dillario, 2023-08-13:
Thank you su much.

dillario, 2023-08-13:
One more question. Can I change the resolution of the camera afterwards?

paukstelis, 2023-08-13:
> One more question. Can I change the resolution of the camera afterwards?

Yes. You must modify the service files (/etc/systemd/system/cam_instancename.service)